OrcaSlicer/OrcaSlicer #5521: Fuzzy skin slicing with identical patterns

cowc1um, 2024-05-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1 beta

### Operating System (OS)

Windows

### OS Version

windows 10

### Additional system information

intel 12900kf, rtx3080

### Printer

Bambu X1C

### How to reproduce

Slice model with fuzzy contour only enabled.

### Actual results

When sliced the first time, some models exhibit contigious areas where the fuzzy skin generated are identical and are aligned, resulting in an area where there is no fuzzy randomness. It seems to happen mostly on cylindrical surfaces
[image]

### Expected results

The fuzzy skin shouldnt align between layers at all and should be completely random. the issue can be remedied by disabling fuzzy skin after the first slice, and re-enabling it and reslicing it, resulting in truely random fuzzi
[image]
ness

### Project file & Debug log uploads

[Fuzzy skin issue project.zip](https://github.com/user-attachments/files/15515647/Fuzzy.skin.issue.project.zip)
First attempt at slicing will show the issue. reslicing the file after disabling and reabling fuzzy will solve the issue

### Checklist of files to include

- [ ] Log file
- [X] Project file

SeaRyanC, 2024-06-10:
Also reported at Bambu https://github.com/bambulab/BambuStudio/issues/4253

This repros pretty obviously with just a cube. I scaled it up on the Z Axis and you can see the patterns readily

[image]

SeaRyanC, 2024-06-10:
I think I have this one figured out (if someone wants to send a PR please do, I don't have time to set up a dev environment at the moment)

Every other function in libslic3r that uses `rand` [ensures it calls `srand(time(NULL));`](https://github.com/SoftFever/OrcaSlicer/blob/aa692b815ea01e84d93c88a8467c6ebdfa5615c9/src/libslic3r/Fill/Lightning/Generator.cpp#L33-L36) first, but `fuzzy_polygon` doesn't ([see here](https://github.com/SoftFever/OrcaSlicer/blob/aa692b815ea01e84d93c88a8467c6ebdfa5615c9/src/libslic3r/PerimeterGenerator.cpp#L68))

Layers get batched out to each CPU which gets a thread-local random seed with the default of `srand(1)`, which means that every worker thread produces the same sequence of noise, leading to the visible patterns. This also explains why the "repeat" workaround mostly works: Eventually the worker thread has processed a slightly different number of layers, meaning it's in a different place in the same sequence of `rand` results.

The easiest way to demonstrate this is that if you restart Orca and re-slice the same cube, the *same* random noise gets applied, but in slightly different order depending on how the threads get scheduled, which is exactly what you see:
[image]

Calling `srand(time(NULL))` is probably not a good fix, since every worker thread is likely to observe the same `time(NULL)` return value and just demonstrate the same problem. [It sounds like you can do something like this](https://stackoverflow.com/questions/15809442/how-to-seed-random-in-different-threads#comment22518144_15809442) to use the thread ID to initialize the RNG, which should guarantee each worker thread gets a different RNG sequence
```cpp
srand(std::hash<std::thread::id>()(std::this_thread::get_id()));
```
But I haven't tried this

closed 2024-06-15 by SoftFever