OrcaSlicer/OrcaSlicer #5523: STP file reader turns concave/hole geometry to convex

spamwax, 2024-05-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Creality CR-6

### How to reproduce

1. Import the attached STEP files in Orca
2. Inspect the parts

[Block-with-spherical-holes.zip](https://github.com/user-attachments/files/15516989/Ball-0.95.zip)

### Actual results

Spheres/holes created using "Revolve" feature of CAD software are randomly(?) presented as protruded solid spheres.

[image]
[image]

### Expected results

There should be 2 carved-out spheres in each part.

[image]

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15516967/log.zip)

[step-file-bug.zip](https://github.com/user-attachments/files/15516975/step-file-bug.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

1. I have opened the attached STEP files in other viewers such as *Open STEP Viewer* and they are rendered correctly.
2. This can also be reproduced in PrusaSlicer (v2.7.4)

foreachthing, 2024-06-01:
Funky STEP files you got there....
I opened them in SolidWorks (looked as you showed) and exported those as STEP again and it worked as expected:

[image]

What CAD have you been using?

spamwax, 2024-06-02:
> Funky STEP files you got there.... I opened them in SolidWorks (looked as you showed) and exported 
> What CAD have you been using?
They are exported by AutoDesk Fusion 360. As mentioned, all of online STEP viewers I tried (plus *Open STEP Viewer* software on Windows) render the funky files correctly.

foreachthing, 2024-06-02:
I agree with you that Orca can improve on the STEP import. I said "funky" because ir looks good in SolidWorks and "inverted" in Orca...

foreachthing, 2024-09-01:
Might be related:  https://github.com/SoftFever/OrcaSlicer/issues/6560

marcosikkens, 2024-09-02:
I closed my issue because it has the same problem as this one. So the issue still persists.

closed 2024-12-09 by github-actions[bot]