OrcaSlicer/OrcaSlicer #5524: Orca Slicer 2.1 Beta crashing after startup

seacred, 2024-05-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU: AMD Ryzen 5 5600 6-Core Processor 3.50 GHz
Board: B550 AORUS ELITE V2
RAM: 32GB
GPU: NVIDIA GeForce RTX3060

### Printer

Voron 2.4, Micorn+, Tridex

### How to reproduce

1. Open Orcaslicer
2. Wait due to startup
3. App crashing
4. ???
5. sad

### Actual results

unable to privide due to circumstances

### Expected results

App working

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15517134/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

SHKinsem, 2024-06-01:
Same problem here, could not launch the app

RatDad1994, 2024-06-01:
Same issue here, updated from an older version to 2.1, near immediate crash on start up.

brave-ulysses, 2024-06-01:
same here 2.0.0 working

ls -lart /opt/bin/OrcaSlicer_Linux_V2.1.0-beta.AppImage 
-rwxrwxr-x 1 svt svt 116425920 May 30 09:02 /opt/bin/OrcaSlicer_Linux_V2.1.0-beta.AppImage

[debug_Sat_Jun_01_15_05_34_130471.log.0.txt](https://github.com/user-attachments/files/15523209/debug_Sat_Jun_01_15_05_34_130471.log.0.txt)

maleldil, 2024-06-07:
I'm running into the same issue on Windows 11 Version 22H2 Build 22621.3672. Windows Event Viewer shows the following information:

```
Faulting application name: orca-slicer.exe, version: 1.0.0.0, time stamp: 0x66077eb7
Faulting module name: d3d11.dll, version: 10.0.22621.3672, time stamp: 0x4ac6f2bf
Exception code: 0xc0000005
Fault offset: 0x000000000007da92
Faulting process id: 0x0x81840
Faulting application start time: 0x0x1DAB902B368C8BD
Faulting application path: C:\Program Files\OrcaSlicer\orca-slicer.exe
Faulting module path: C:\Windows\SYSTEM32\d3d11.dll
Report Id: f6bc0161-e93d-4bb0-8dec-d2494e1a733d
Faulting package full name: 
Faulting package-relative application ID: 
```

The crash log in %APP_DATA%\Roaming\OrcaSlicer contains the following:

```
Exceptions:

Exception Code :c0000005 ACCESS_VIOLATION

Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x0 
Context :000000F7B4DFA340 
ContextFlag : 0x10005f, EFlags: 0x10202 
Fault address:  0x824ADA92 0x1:0x7CA92 C:\Windows\SYSTEM32\d3d11.dll

Show CallStack:
7ffa824ada92:D3D11CreateDeviceForD3D12 [Unknown file][-1]
7ffa824af026:D3D11On12CreateDevice [Unknown file][-1]
180008c83:Unknown [Unknown file][-1]
18001497f:Unknown [Unknown file][-1]
1800416e6:InitPreview [Unknown file][-1]
18004527c:InitPreview [Unknown file][-1]
1800453ff:InitPreview [Unknown file][-1]
18003ae6b:UninstallRTSSHook [Unknown file][-1]
7ff9da489bfd:DrvSetCallbackProcs [Unknown file][-1]
7ff9da48be0c:DrvSetCallbackProcs [Unknown file][-1]
7ff9da4282b6:DrvSetCallbackProcs [Unknown file][-1]
7ff9da26edd1:DrvSetCallbackProcs [Unknown file][-1]
7ff9da1e8fe6:DllMain [Unknown file][-1]
7ff9da1e6859:DllMain [Unknown file][-1]
7ff9da1eb60f:DllMain [Unknown file][-1]
7ff9db6c5292:eglGetProcAddress [Unknown file][-1]
7ffa88aa257d:BaseThreadInitThunk [Unknown file][-1]
7ffa8a24aa48:RtlUserThreadStart [Unknown file][-1]

Logical Address:
0x824ADA92 0x1:0x7CA92 C:\Windows\SYSTEM32\d3d11.dll
0x824AF026 0x1:0x7E026 C:\Windows\SYSTEM32\d3d11.dll
0x80008C83 0x1:0x7C83 C:\Program Files (x86)\RivaTuner Statistics Server\RTSSHooks64.dll
0x8001497F 0x1:0x1397F C:\Program Files (x86)\RivaTuner Statistics Server\RTSSHooks64.dll
0x800416E6 0x1:0x406E6 C:\Program Files (x86)\RivaTuner Statistics Server\RTSSHooks64.dll
0x8004527C 0x1:0x4427C C:\Program Files (x86)\RivaTuner Statistics Server\RTSSHooks64.dll
0x800453FF 0x1:0x443FF C:\Program Files (x86)\RivaTuner Statistics Server\RTSSHooks64.dll
0x8003AE6B 0x1:0x39E6B C:\Program Files (x86)\RivaTuner Statistics Server\RTSSHooks64.dll
0xDA489BFD 0x1:0xB18BFD C:\Windows\System32\DriverStore\FileRepository\u0402593.inf_amd64_66758692e35a2906\B402774\atio6axx.dll
0xDA48BE0C 0x1:0xB1AE0C C:\Windows\System32\DriverStore\FileRepository\u0402593.inf_amd64_66758692e35a2906\B402774\atio6axx.dll
0xDA4282B6 0x1:0xAB72B6 C:\Windows\System32\DriverStore\FileRepository\u0402593.inf_amd64_66758692e35a2906\B402774\atio6axx.dll
0xDA26EDD1 0x1:0x8FDDD1 C:\Windows\System32\DriverStore\FileRepository\u0402593.inf_amd64_66758692e35a2906\B402774\atio6axx.dll
0xDA1E8FE6 0x1:0x877FE6 C:\Windows\System32\DriverStore\FileRepository\u0402593.inf_amd64_66758692e35a2906\B402774\atio6axx.dll
0xDA1E6859 0x1:0x875859 C:\Windows\System32\DriverStore\FileRepository\u0402593.inf_amd64_66758692e35a2906\B402774\atio6axx.dll
0xDA1EB60F 0x1:0x87A60F C:\Windows\System32\DriverStore\FileRepository\u0402593.inf_amd64_66758692e35a2906\B402774\atio6axx.dll
0xDB6C5292 0x1:0x1D54292 C:\Windows\System32\DriverStore\FileRepository\u0402593.inf_amd64_66758692e35a2906\B402774\atio6axx.dll
0x88AA257D 0x1:0x1157D C:\Windows\System32\KERNEL32.DLL
0x8A24AA48 0x1:0x59A48 C:\Windows\SYSTEM32\ntdll.dll
```

I uninstalled 2.1.0-beta and reinstalled 2.0.0 and it worked again. Hope this information can be useful.

seacred, 2024-06-10:
Fixed the issue on my machine via uninstalling Version 2.0.0 and installing 2.1.0 afterwards.
Letting the v2.1.0 installer handle the uninstall of 2.0.0 seems to mess it up.

closed 2024-06-10 by seacred