OrcaSlicer/OrcaSlicer #5525: Thumbnail broken in 2.1.0 beta

adiyahu, 2024-05-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Flsun V400

### How to reproduce

1. Slice
2. Print
3. Check thumbnai in web interface or pad interface.

### Actual results

The print is normal, the thumbnail is empty both in web interface and on the pad interface. It was all fine before update to 2.1.0 beta

### Expected results

Thumbnail should be displayed

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15517314/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

[image]

Comar2557, 2024-05-31:
This problem is also occurring for me

Arakon, 2024-06-01:
It's showing on both latest Klipper and Bambulab for me.

igiannakas, 2024-06-02:
Hm cannot reproduce - I’ve set the thumbnail resolution to 400x400. Try that and let us know?

adiyahu, 2024-06-02:
> Hm cannot reproduce - I’ve set the thumbnail resolution to 400x400. Try that and let us know?

Nope, not working. The thing was that, I tried on a multi-plate project, for some plate, the thumbnail showing weird part from neigbour plate, but not all parts on that plate, in most cases, the thumbnail is still empty.

igiannakas, 2024-06-02:
There’s some thing odd going on here - see below a print I’m starting just now 🤔
[image]

is klipper version selected in your printer type? Also do you have object processing (exclude object) enabled in moonraker?

adiyahu, 2024-06-02:
> is klipper version selected in your printer type? 
I'm using the latest official klipper, and it's all fine when I switch back to 2.0.0 to slice. So I think it has nothing to do with the klipper (though I know FLsun's stock klipper is out-dated, that I won't show thumbnail for Orca)

>Also do you have object processing (exclude object) enabled in moonraker?
Yes I enabled it. But again, it's all fine with v2.0.0, so must be something wrong with 2.1.0 beta?

igiannakas, 2024-06-02:
No I mean is klipper selected in orca as the printer type?

tue above screenshot is with latest klipper and orca 2.1 beta so there is something else going on here

adiyahu, 2024-06-02:
> No I mean is klipper selected in orca as the printer type?

Sure, of course klipper is selected. Though my previous configs are unchanged, I did double check all related options.
 
> tue above screenshot is with latest klipper and orca 2.1 beta so there is something else going on here

Since it's a beta, I think it might be a bug uncommon but on some occasions. I see someone else has the same issue, so I'm not alone

adiyahu, 2024-06-02:
[image]
This is the thumbnail generated in the gcode file. Surely there is something very odd here

igiannakas, 2024-06-02:
Yeap that is not right. Can you upload your project file (.3mf compressed in a zip)? It will help reproducing the issue - the OP only contains the log file, not the project file.

adiyahu, 2024-06-02:
> Yeap that is not right. Can you upload your project file (.3mf compressed in a zip)? It will help reproducing the issue - the OP only contains the log file, not the project file.
[2.1test.3mf.zip](https://github.com/user-attachments/files/15524925/2.1test.3mf.zip)
sure, here it is

igiannakas, 2024-06-02:
Issue confirmed - **Orca fails to export thumbnail on printers with a circular bed - if you use a rectangular bed it exports the image successfully**. 

@SoftFever this is beyond my skill level I'm afraid :( Possibly some change introduced to support the larger plate sizes or similar broke export of thumbnails for circular bed printers.

igiannakas, 2024-06-02:
More specifically, the render_thumbnail_framebuffer function returns all \0's in the pixels rendered. 

[image]

adiyahu, 2024-06-02:
> Issue confirmed - **Orca fails to export thumbnail on printers with a circular bed - if you use a rectangular bed it exports the image successfully**.
>
> @SoftFever this is beyond my skill level I'm [...]

Then it is a new issue comes with 2.1.0.
Speaking of the support to the circular bed, the automatic parts layout on bed always failed in circular bed, as long as I'm using Orca. Is it possible to look into this matter too? @SoftFever

igiannakas, 2024-06-02:
@adiyahu Not sure whether this is new or not - have not checked against the previous release. Does it happen if you slice with the 2.0 release version? It would help triangulate whether this issue was introduced now or earlier.

adiyahu, 2024-06-02:
> @adiyahu Not sure whether this is new or not - have not checked against the previous release. Does it happen if you slice with the 2.0 release version? It would help triangulate whether this issue was introduced now or earlier.

There is no such issue in previous versions. I have both version installed, 2.0 and 2.1beta, same settings and config, different outcomes

SoftFever, 2024-06-03:
I can repro it as well.

closed 2024-06-04 by SoftFever

SammyHC2, 2024-06-10:
I put this configuration and it came back Thumbnail
[image]

adiyahu, 2024-06-11:
> I put this configuration and it came back Thumbnail ![Thumbnail](https://private-user-images.githubusercontent.com/172323104/338343618-1b4dc1fe-fafd-4c67-870f-0f8db0abe0b7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgwNjQwNzIsIm5iZiI6MTcxODA2Mzc3MiwicGF0aCI6Ii8xNzIzMjMxMDQvMzM4MzQzNjE4LTFiNGRjMWZlLWZhZmQtNGM2Ny04NzBmLTBmOGRiMGFiZTBiNy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYxMFQyMzU2MTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NTNkY2M0NmUzOTQ4NDRkMDM3ZTIxYmUxMGE1NTMzNTJlMTQ0OTg5NTlkNDEzNGRkY2FjYTc4NzgxZWU3YWM2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.YuNS_LH6bl1NmhJsTzMu5oP_Ca3nG9xT-pBVWq-I3bk)

The thumbnail apears, but the content isn't correct in my case. Still not a temporary solution. I have to wait for the next update

adiyahu, 2024-06-11:
> Can you download the latest nightly build and give it a try?

Just did, and the problem fixed. Thanks!

patg84, 2024-08-16:
> I put this configuration and it came back Thumbnail [...]

Had to do this in v2.1.1 to get thumbnails to work.

commit ae07cf98 (2024-06-04): fix a regression that thumbnail was not rendered properly for delta printers

Fixes #5525