OrcaSlicer/OrcaSlicer #5526: Bambu long retract when cut was ported over in custom g code, but not the checkbox to enable it

chris99336, 2024-05-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-Beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

N/A

### Printer

Bambu X1C

### How to reproduce

Easy to reproduce. The printer profile simply has no option to enable the flag for the filament change gcode to perform a long retract on cut

### Actual results

There is nothing for me to click

### Expected results

There should be a box to turn on the long retract.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I know this is a beta version, and things get missed. Bambu added the long retract on cut to reduce purge waste, I was excited to see that it was ported to Orca Slicer. However, in the extruder options in Bambu Studio, there is a checkbox to enable it that did not make it into Orca Slicer. The custom gcode that looks for that flag exists. There is just no way to set the flag.

Alanoll, 2024-06-02:
The option is filament specific, not extruder.

[image]

SoftFever, 2024-06-03:
As mentioned by @Alanoll , this flag is in filament settings instead of extruder settings.

chris99336, 2024-06-05:
I totally missed that. Thank you both @Alanoll @SoftFever

closed 2024-06-05 by chris99336

tbonge, 2024-07-15:
Will this be implemented on the A1?

Arakon, 2024-12-10:
The A1 and A1 Mini have received a firmware update [today](https://wiki.bambulab.com/en/a1/manual/a1-firmware-release-history) to support this now. Please implement the option for them as well.

aeschrud, 2025-07-19:
Sorry to bother but any chance this can be implemented vor the Creality printers as well? I'm a K2 plus owner and the longer nozzle really needs this feature :-)