OrcaSlicer/OrcaSlicer #5529: Open Cura/Prusa doesn't open in Orcaslicer

Avpman2, 2024-05-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 beta

### Operating System (OS)

Windows

### OS Version

Win 11

### Printer

Flashforge

### How to reproduce

Go to the Printables, Maker, or Thingiverse and click the icon to open in Prusaslicer, Bamboo, or Cura  The settings in Orcaslicer are set correctly:

[image]

### Actual results

The message to open Prusa, Bamboo, or Cura appears, saying the respective program is not installed,

### Expected results

Orcaslicer should run, and it does not.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2024-06-03:
can you check if the the association is correct? You can check the full path of the associated application by hovering on the text

Avpman2, 2024-06-03:
It turns out the prompt from the model's website says Cura, Prusa, etc., and you just have to accept that prompt before the slicer prompt appears and specifies Orcaslicer. So it's not a bug; just needs clarification in the docs. Example (confusing) prompt from the model's website:

[image]

JimmysDev, 2024-06-04:
I am currently experiencing this with Thingiverse. Running 2.1.0 on macOS. If I click the link it tells me I need to download Cura. If I say "Download Cura" it takes me to the download page. If I say "Okay" it closes the dialogue. For the life of me I can't find the settings in the screenshot originally posted, so I don't know if they are enabled or not.

Avpman2, 2024-06-04:
In Orcaslicer hit Ctrl-P for the preferences page. Scroll down until you get to the settings I posted yesterday above.

JimmysDev, 2024-06-04:
> In Orcaslicer hit Ctrl-P for the preferences page. Scroll down until you get to the settings I posted yesterday above.

On macOS it's Cmd+, and this is all I see. There isn't a scroll bar to go further down:
[image]

Definitely using the beta:
[image]

Avpman2, 2024-06-04:
Ahhh, per the release notes, it only works on Windows.

SoftFever, 2024-06-05:
Yeah, Mac is not supported yet.
I haven’t found a way to dynamically register Orca as the default application to handle web link requests. And I don’t want to automatically associate these web links without the user’s explicit consent.

Kurre, 2024-06-24:
> Yeah, Mac is not supported yet. I haven’t found a way to dynamically register Orca as the default application to handle web link requests. And I don’t want to automatically associate these web links without the user’s explicit consent.

As a side note, macOS users can use e.g. https://github.com/Lord-Kamina/SwiftDefaultApps and manually set those link associations. Tested with all three ones:

[image]
[image]
[image]

of course this is just a work-around until macOS associations are fully supported by Orca, but for me having these associations is a big deal 🤩 👍

ender3, 2024-07-17:
@Kurre I have set all three associations, but I receive "malformed url" responses from OrcaSlicer. Did you do anything else to make it work?

Kurre, 2024-07-29:
> @Kurre I have set all three associations, but I receive "malformed url" responses from OrcaSlicer. Did you do anything else to make it work?

Sorry it took so long to answer.

No, I don't recall doing anything else to make it work 🤔

3DCoded, 2024-08-25:
> As a side note, macOS users can use e.g. https://github.com/Lord-Kamina/SwiftDefaultApps and manually set those link associations. Tested with all three ones: [...]

Thank you for this! I have wanted this feature on my Mac for a while. This should be added to the documentation as a workaround for Mac.

closed 2024-12-01 by github-actions[bot]