OrcaSlicer/OrcaSlicer #5531: Orca slicer hangs printing this file.

Oscilllator, 2024-06-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer_Linux_V2.0.0.AppImage

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04.3

### Printer

Bambu P1P

### How to reproduce

Both the zip and .stl versions when sliced with the default 0.4mm profile of the p1p. When I press 'print' the program hangs. I don't think it is hanging just to transfer the file, I left it for 2 hours and nothing happened.
[image]

### Actual results

Program stops responding

### Expected results

printer should start printing.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15520417/log.zip)
[Part Studio 1.zip](https://github.com/user-attachments/files/15520414/Part.Studio.1.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2024-06-03:
does it happen every time?

Oscilllator, 2024-06-04:
Yes, every time. I am attempting to split the model right now to print it in halves to get around the problem.

SoftFever, 2024-06-05:
Might be a network issue, the log didn't show errors.

Oscilllator, 2024-06-06:
Interesting, on startup I have been getting the message:
```
use system SSL certificate: /etc/ssl/certs/ca-certificates.crt
To manually specify the system certificate store, set the SSL_CERT_FILE environment variable to the correct CA and
restart the application
```
[image]

I wonder if this is related.

KollegeX, 2024-06-13:
Probably related to  #2326 
Experience the same thing on every large gcode with 2.0 and 2.1 beta
Using the Appimage of 1.8 allowed me to print

olehmykhalchenko, 2024-06-13:
Its related to bambu network driver. Works good on Orca Slicer v1.8.1, also works on bambu studio. All newer versions of orca slicer are broken.

olehmykhalchenko, 2024-06-14:
Also important to mention this problem is not reproduced with Klipper integration. So its 100% bambu network driver.

disconn3ct, 2024-06-20:
It is happening here in 2.0 appimage. The app is hammering a single cpu at 100% and strace shows it `read()`ing all of the gcode as ascii.

I wonder if this is a stupid way to copy the gcode across the licensing barrier or something.

```
read(86, "982.376\nG1 X198.363 Y103.503 E.0"..., 8191) = 8191
read(86, "F12000\nG1 X126.81 Y67.289 E.0101"..., 8191) = 8191
read(86, "343.004\nG1 X174.396 Y72.967 E.01"..., 8191) = 8191
read(86, "3\nG1 X200.823 Y110.183 E.00173\nG"..., 8191) = 8191
read(86, "8.264 Y159.591 E.01463\nG1 F12000"..., 8191) = 8191
read(86, "00828\nG1 X146.381 Y180.343 E.011"..., 8191) = 8191
```
FD 86 (and 85 for some reason) is `/tmp/orcaslicer_model/Thu_Jun_20/11_39_58#2836763#269/Metadata/.2870340.0.gcode` and it is 75MB.

This happens at the "pick a printer" dialog of the "print plate" button, BEFORE the "send to printer" was clicked. I'm running with an AMS, and the slicer is connected (device page) but it hangs before the filament selection on the dialog. This particular model is fuzzy skin.

olehmykhalchenko, 2024-06-24:
Also not working in new 2.1.0 release

af7v, 2024-06-27:
I can confirm that v2.1.1 hangs on large gcode files when trying to print. Appimage on Linux. Smaller files work fine.
In my case I'm trying a stacked multiboard sending to a BambuLab X1C

Highcooley, 2024-07-01:
> I can confirm that v2.1.1 hangs on large gcode files when trying to print. Appimage on Linux. Smaller files work fine. In my case I'm trying a stacked multiboard sending to a BambuLab X1C
> 
> Opened source and sliced to a 15.3MB file in Bambu Studio 1.9.3.50 and was able to send to the printer.

Same finding in v2.1.1 running as appimage on Linux: larger files (did not check the size limit) hang, while they can be printed with Bambulab easily. Also sending to SD card has the same issue.

Highcooley, 2024-07-01:
Maybe the ticket should be renamed "Orca slicer hangs printing large files", to make it more generic as the problem seems to be.

bhaal2, 2024-08-01:
Can confirm this is still an issue in current nightly build as of the 01/08/24 (AEST)

nsfilho, 2024-08-16:
The problem persists in version 2.2.0-dev. I'm using Debian 12, and if I reduce the G-code, I can select the printer and send it. However, with large G-code files, it freezes.

Is there any way I can assist in resolving this bug? If a developer could provide some guidance on the possible cause, I would be happy to investigate and try to help with a PR.

eous, 2024-09-10:
So here is a bit that I don't see yet. I think its continuously re-reading the gcode because the printer keeps rejecting whatever handshake its trying to do.

(Modified to remove some sensitive bits)
```13:57:23.929463 IP (tos 0xc0, ttl 64, id 33631, offset 0, flags [none], proto ICMP (1), length 481)
     > : ICMP  udp port 16147 unreachable, length 461
	IP (tos 0x0, ttl 64, id 39621, offset 0, flags [DF], proto UDP (17), length 453)
     > : [udp sum ok] UDP, length 425
	0x0000:  c0f5 354e caa4 107c 610c 04c1 0800 45c0  ..5N...|a.....E.
	0x0010:  01e1 835f 0000 4001 7332 c0a8 0004 c0a8  ..._..@.s2......
	0x0020:  0076 0303 808a 0000 0000 4500 01c5 9ac5  .v........E.....
	0x0030:  4000 4011 1c98 c0a8 0076 c0a8 0004 07e5  @.@......v......
	0x0040:  3f13 01b1 3418 4854 5450 2f31 2e31 2032  ?...4.HTTP/1.1.2
	0x0050:  3030 204f 4b0d 0a53 6572 7665 723a 2055  00.OK..Server:.U
	0x0060:  506e 502f 312e 300d 0a44 6174 653a 2054  PnP/1.0..Date:.T
	0x0070:  7565 2c20 3130 2053 6570 2032 3032 3420  ue,.10.Sep.2024.
	0x00b0:  616d 6275 6c61 622d 636f 6d3a 6465 7669  ambulab-com:devi
	0x00c0:  6365 3a33 6470 7269 6e74 6572 3a31 0d0a  ce:3dprinter:1..
	0x00f0:  6865 2d43 6f6e 7472 6f6c 3a20 6d61 782d  he-Control:.max-
	0x0100:  6167 653d 3138 3030 0d0a 4465 764d 6f64  age=1800..DevMod
	0x0110:  656c 2e62 616d 6275 2e63 6f6d 3a20 424c  el.bambu.com:.BL
	0x0120:  2d50 3030 310d 0a44 6576 4e61 6d65 2e62  -P001..DevName.b
	0x0130:  616d 6275 2e63 6f6d 3a20 5831 430d 0a44  ambu.com:.X1C..D
	0x0140:  6576 5369 676e 616c 2e62 616d 6275 2e63  evSignal.bambu.c
	0x0150:  6f6d 3a20 2d33 340d 0a44 6576 436f 6e6e  om:.-34..DevConn
	0x0160:  6563 742e 6261 6d62 752e 636f 6d3a 2063  ect.bambu.com:.c
	0x0170:  6c6f 7564 0d0a 4465 7642 696e 642e 6261  loud..DevBind.ba
	0x0180:  6d62 752e 636f 6d3a 206f 6363 7570 6965  mbu.com:.occupie
	0x0190:  640d 0a44 6576 7365 636c 696e 6b2e 6261  d..Devseclink.ba
	0x01a0:  6d62 752e 636f 6d3a 2073 6563 7572 650d  mbu.com:.secure.
	0x01b0:  0a44 6576 496e 662e 6261 6d62 752e 636f  .DevInf.bambu.co
	0x01c0:  6d3a 2077 6c61 6e30 0d0a 4465 7656 6572  m:.wlan0..DevVer
	0x01d0:  7369 6f6e 2e62 616d 6275 2e63 6f6d 3a20  sion.bambu.com:.
	0x01e0:  3031 2e30 382e 3032 2e30 300d 0a0d 0a    01.08.02.00....
13:57:23.981469 IP (tos 0x0, ttl 64, id 43117, offset 0, flags [DF], proto UDP (17), length 109)
     > : [udp sum ok] UDP, length 81
13:57:23.991535 IP (tos 0x0, ttl 64, id 43122, offset 0, flags [DF], proto UDP (17), length 101)
     > : [udp sum ok] UDP, length 73```

JacobTheEldest, 2024-11-05:
Same issue as https://github.com/SoftFever/OrcaSlicer/issues/5273?

su-sd, 2024-12-03:
> Might be a network issue, the log didn't show errors.

Definitely not a network problem, since it only appears while printing, not when sending the file to the sd-card.
I tried it in LAN mode and over bambu cloud multiple times, both had the same problem.
Bambu Studio also hangs on my file.

eous, 2024-12-03:
Fwiw with the current bambu network plugin and head of tree this is no longer an issue for me.

astfgl, 2025-01-19:
I can confirm that the current main codebase will now upload a large file (>6MB) to the printer using the "Send" function, however the "Print plate" function gives an FTP error.

Debian Unstable
Github build instructions resulting in build/OrcaSlicer_Linux_V2.3.0-dev.AppImage
Bambu Lab P1S 01.07.00.00
libbambu_networking.so Aug 28 21:22 6319464

Error code : -4020
Error desc : Failed to upload the print file to FTP. Please check the network status and try again.
Extra into  : [ftp code]: 9
                     [reason]: Access denied to remote resource:Server denied you to change to the given directory

Note that the upload using the "Send" function works correctly, however I cannot find any way of remotely starting the print job in LAN mode.

It appears that the FTP session is over an encrypted connection. Are there any debug settings I can turn on to assist with this issue?

3vi1, 2025-01-23:
I started playing with LAN mode last weekend and I have also encountered this issue because I tend to work with .step files from Onshape, which are rather large at the default import settings. 

What I've seen in my testing is that everything responds fine until somewhere between 300k and 400k triangles.  At 400k+ triangles, the print dialog becomes unresponsive as soon as you bring it up with the "Print plate" button.

When the dialog becomes unresponsive at 400k+ triangles, you _can_ close it with the close button in the top right, you just have to wait a minute for it to finally recognize the click.  If at that point you delete some objects such that you're back down to ~300k triangles, you can re-open the print dialog and it will respond very snappy again.  So... there's some specific demarcation point in model complexity where things immediately get super slow.

It's consistent and repeatable, and it doesn't have anything to do with the network transport itself:  I can send the same 400k+ triangle files, and much larger ones, to the printer with Bambu Studio (also compiled from source) with no issue.  Also, OrcaSlicer has no problem accessing the printer for status/camera.

OS:  Linux (Ubuntu 25.04 Plucky Puffin beta)
OrcaSlicer version:  2.3.0-dev - built from current git each day I test.

3vi1, 2025-01-23:
It turns out that the amount of support material needed for the objects is also a factor. 

I was able to get it to still respond at 400k triangles in a project that had fewer supports. Here's a video show that project as being responsive at 4 objects (100k each) and then the print dialog showing the problem when I clone in one more object: https://youtu.be/l062I3wq-9U

Beep6581, 2025-02-23:
Still valid in v2.3.0-beta.
Duplicate of #3893

dmach, 2025-03-09:
Also reproducible in v2.3.0-beta2.
Bambu A1, happens before hitting "Send".

3vi1, 2025-03-15:
I see the cause and have reported a fix in https://github.com/SoftFever/OrcaSlicer/issues/3893#issuecomment-2726639245.

I didn't submit a pull request because @SoftFever or one of the other regulars might want to solve it in a slightly different manner, but I can create one if they don't have time to work on it.

jzobac, 2025-06-14:
Is it solved?

3vi1, 2025-06-14:
Yes.  This was fixed in March, but I don't have the ability to close this issue.  https://github.com/SoftFever/OrcaSlicer/pull/8925