OrcaSlicer/OrcaSlicer #5534: Text size inconsistency

Eldenroot, 2024-06-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0beta

### Operating System (OS)

Windows

### OS Version

Win 11

### Printer

A1 with AMS

### How to reproduce

[image]

### Actual results

[image]

### Expected results

Enable AMs should use same font size and be aligned

### Project file & Debug log uploads

---

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Eldenroot, 2024-06-01:
There is another - plate edit parameter

[image]

yw4z, 2024-06-15:
i will look a solution for this but i dont think there is an easy fix.

closed 2024-09-21 by github-actions[bot]