OrcaSlicer/OrcaSlicer #5536: Transfer settings popup - design improvents

Eldenroot, 2024-06-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Would be nice to adapt some changes from latets Bambustudio into Orca

[image]

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It is quite misleading and confusing, at least color and title should be added

SoftFever, 2024-06-03:
Can you clarify which part is misleading?
I keep it consistent with PrusaSlicer and I find it's more clear/precise in current design

Eldenroot, 2024-06-03:
https://github.com/SoftFever/OrcaSlicer/issues/4422#issuecomment-2144307140

SoftFever, 2024-06-03:
Hmm, but that's in 2.0.0, in 2.1.0-beta, I have changed that.
Anyway let's continue the  discussion there instead of creating a duplicated ticket

closed 2024-06-03 by SoftFever