OrcaSlicer/OrcaSlicer #5537: Add "Snap to orthographic view" in the plate view like Blender alt-orbit

J3LS, 2024-06-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I wouldn't call this a problem but it would be, I think, a great enhancement with no downside.

At the moment the only views available are pseudo-3d which makes looking along edges a little awkward.

When left clicking and orbiting the plate view it would be really useful to easily see orthogonal views from top, front, back, left and right

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Implement the same intuitive function as in Blenders 3D view. In Blender orbit is holding the mouse wheel down. Holding Alt before clicking the mouse-wheel snaps the view to orthogonal quarter and half quarter views. 

In Orca we would only need top, left, right, front, back. Where orbit in Orca is left click-orbit, snap to orthogonal could be alt-left click orbit.

This would help greatly for looking along edges and for alignment etc.

[image]

### Describe alternatives you've considered

It is possible to to this in Orca's plate view but it's clunky and you can only see one edge at a time. I can't see a good alternative.

### Additional context

This would have no negative impact as only holding alt while left clicking would activate the feature

Noisyfox, 2024-06-04:
[image]
For now you could manually switching between the two views.
Also you can clik the 3d navigator to switch to top, left, right, front, back etc. views quickly, or use the view manu.

J3LS, 2024-06-04:
My bad! I didn't realise that existed. No real point in my suggestion then. I'll RTFM next time. Thanks

closed 2024-06-04 by J3LS