OrcaSlicer/OrcaSlicer #5539: Turn off Lamp after print finishes on Bambulab printers (and other applicable printers)

zwamdurkel, 2024-06-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm always frustrated when I print something and the lamp in my P1S stays on after the print is finished. I did not turn on the lamp, the printer did, so it should also turn it off after the print.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

It would be amazing if the user could configure whether to turn on or off the lamp when you print and whether to turn on or off the print after it is finished.

A way to present these options could be as follows:

Lamp while printing (on/off)
Lamp after print finished (on/off)

### Describe alternatives you've considered

I have briefly looked into this issue and there does not seem to be a gcode command to do this. If I had found a gcode command, I would've added it to the custom start and end code. 

### Additional context

This seems to be a common frustration with bambulab printers. There is an issue on their github that has been open for ages with no resolution in sight https://github.com/bambulab/BambuStudio/issues/1957

efab2022, 2024-07-03:
+1 please

closed 2024-10-09 by github-actions[bot]

treyus30, 2025-06-03:
Reviving as this needs to be implemented.

That and ideally a several-minute delay to turning the chamber fan off (or immediately after print is done). 

Fans, and LEDs especially, have a lifespan. These two items will be left on sometimes for days if I'm not back at the house to manually turn them off.