OrcaSlicer/OrcaSlicer #5550: Access violation when trying to slice after switching to another preset and back again.

akoolenbourke, 2024-06-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 Beta

### Operating System (OS)

Windows

### OS Version

WIndows 10

### Additional system information

AMD 3900x
32GB RAM

### Printer

Creality CR10 Smart Pro

### How to reproduce

Not sure how anyone else can reproduce but I made a 1mm nozzle preset the other day and today I switched to it and back to my usual 0.4. Then I went to slice and got the error.

I can no longer slice with any of my profiles without this error.

### Actual results

Exception as outlined in the screenshot

### Expected results

Slicing works.

### Project file & Debug log uploads

[AccessViolationProject.zip](https://github.com/user-attachments/files/15523851/AccessViolationProject.zip)
[log.zip](https://github.com/user-attachments/files/15523852/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I upgraded to 2.1.0-beta to see if it'd fix the issue but the same problem happened.
Now I cannot slice with any of my profiles at all.

One thing of NOTE is that I have built Orca Slicer myself and ran the build too in order to do make some changes. When I switch between running the official release and my development build, Orca always redownloads the Bambu Network plugin. That's fine, I can handle that but I have also noticed that more often than not when I first go back to running Orca release, my user presets don't load. I then quit Orca, start it again and they are there. 

I have attached my log files from today. Not all saw this issue but I'm not sure which one it started on.
[image]

SoftFever, 2024-06-04:
I checked the project file, the small_area_infill_flow_compensation_model are indeed wrong there.
But nevertheless, it should prevent slicing as small_area_infill_flow_compensation was not enabled.
The issue should be fixed now.

[image]

closed 2024-06-04 by SoftFever

TheSlashEffect, 2024-06-21:
I've located an issue with the commit addressing this issue; I'm building from master and the value of `m_config.small_area_infill_flow_compensation.value` is not read as true when the feature is enabled, making the feature not activate.
If I revert 9a4429f6037bb87001b78d9fe9f158bd5f1f4a61 the feature works as expected again.
Alternatively, if I move the definition of the config value `small_area_infill_flow_compensation` from `PrintRegionConfig` to `PrintConfig`, the feature works again.
The value was moved to the region config due to [this conversation](https://github.com/SoftFever/OrcaSlicer/pull/3334/commits/44665b39e1a43c29fa2bf9a4d566c89838545969#r1459245343):
How should this be addressed? I'm looking into how to make the current revision work with the parameter in the region config.

commit 9a4429f6 (2024-06-04): Fixed an issue that small_area_infill_compensation_model was initialized even it's not used.
Fixed #5550