OrcaSlicer/OrcaSlicer #5552: display of remaining time and percentage calculation

BUZZ-SEB, 2024-06-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

windows 10

### Printer

Creality ender 3 V3 KE

### How to reproduce

Hi, when I send a piece in orcaslicer with my ender 3 V3 KE printer, the remaining time and the percentage are not displayed on the control panel or on creality cloud. I tested MAINSAIL or FLUIDD, it's the same.
do you have an idea?
Thanks 

### Actual results

[image]
[image]

### Expected results

the total time of the print duration should be displayed along with the remaining time.
the percentage progress bar should change. There are problems when sending from Creality print or Creality cloud.

### Project file & Debug log uploads

[orca pb.zip](https://github.com/user-attachments/files/15524908/orca.pb.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Jaizu, 2024-06-22:
I tried the latest version and I still can't find the cause for this, the crazy this is that I was able to get it working some times.
I will do more testing with the files that showed the progress to see if I can see anything weird or different in the .gcode.

Jaizu, 2024-06-22:
Not sure if this is an issue with orca slicer, tested uploading a gcode file file from mainsail that was generated with Creality Print and it has the same issue.
[image]

BUZZ-SEB, 2024-06-22:
the photo I posted shows an stl that I cut with ORCA.

Jaizu, 2024-06-22:
My point is that this issue has nothing to do with Orca Slicer so it shouldnt be reported here.

BUZZ-SEB, 2024-06-22:
However when I take the same stl with creality print or cura, no problem.

Jaizu, 2024-06-23:
Orca is not the issue, the pic that I showed was stuck at 0% too and it was madre with Creality Print.

Jaizu, 2024-06-30:
To make more clear why this isnt an issue with orca slicer and this should be closed.
I slice something with orca slicer.
If I send the file from mainsail, fluidd or orcaslicer and print from there, the display time and progress wont show.
If I send the file to the print and i hit print from the nebula pad, it will display everything as intended.

closed 2024-10-07 by github-actions[bot]