OrcaSlicer/OrcaSlicer #5554: error 2:-10 when try to see the bambulab cam

thezraner, 2024-06-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

Windows 11 updated

### Printer

Bambulab P1S

### How to reproduce

1. Open OrcaSlicer 2.1.0
2. Go to "Device"
3. Press play for the camera
4. Recieve the error 2:-10

### Actual results

The camera shows all black with the error 2:-10

### Expected results

The camera shows the image from the printer

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15525130/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

msteele999, 2024-06-02:
Confirmed - having the same issue here.

Bambu X1CC, Windows 11

SecretlyaGoodGuy, 2024-06-02:
Bambu P1s, Windows 11

Same Problem here, everything is working fine except the Video.

Bob-Soft, 2024-06-02:
Same problem

PMilstone, 2024-06-03:
Same issue here, unable to view camera through OrcaSlicer. Camera working fine in Bambu Handy.

My P1S just updated to the latest firmware so I suspect that's related.

SecretlyaGoodGuy, 2024-06-03:
@PMilstone my P1S is not updated by now, so it don't need to be related

scorfman, 2024-06-04:
Also experiencing this issue pre and post the latest bambulab software release 1.08. Works fine via BambuStudio and handy.

balp, 2024-06-04:
Initial testing shows: Offical builds: Windows 11, affected, MacOS, Ubuntu not affected.

balp, 2024-06-04:
After running the mac/linux version, and Bambu Studio  1.9.1.66 the issue on Windows 11, can't be reproduced anymore.

balp, 2024-06-04:
Do OrcaSlicer and Bambu Studio share any libraries from the Bambu plugin? I didn't see Orca update reinstall it, but I think the Bambu Update pulled in a new plugin. If I recall correctly the humidity level drop has changed, on the AMS view now it's a clear drop with numbers, don't think I had that when I had issues. I hope this small testing helps locate the problem.

SoftFever, 2024-06-05:
This should be fixed in the upcoming nightly build. 
Unfortunately, my P1S's webcam died, so I can't test it at the moment. 
For anyone encountering this issue, please try the latest nightly build and let me know if it resolves the problem.

closed 2024-06-05 by SoftFever

thezraner, 2024-06-05:
i have tried the portable nightly version, and it's fully working, thx so much!

msteele999, 2024-06-05:
> Yep its now showing the P1S camera on my win11 machine using the 1bb8fad
> build! Thank you SoftFever, great work! I was using my phone for my camera
> view and was getting HOT.

Confirmed nightly build now has camera on X1CC. Note - got an error message
on initial startup that camera was associated with a different build -
allowed it to apply a registry fix and then it worked.

SecretlyaGoodGuy, 2024-06-05:
Yep, works perfectly with the Nightly, Thanks for the fast fix.

Got the same, error message first and then all worked with pressing yes on it and a restart of the Slicer.

commit cb4b84c4 (2024-06-05): Fix an issue that Bambu printer camera is not prompted to register due to the Bambu's registry path changes
Fix #5554