OrcaSlicer/OrcaSlicer #5562: Incorrect Overhangs

butchs38, 2024-06-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

windows 10

### Printer

qidi xmax 3

### How to reproduce

load and slice

### Actual results

[image]

### Expected results

Should not be a overhang line shows up in print

### Project file & Debug log uploads

[fb_l_dt2.zip](https://github.com/user-attachments/files/15531048/fb_l_dt2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]

SoftFever, 2024-06-03:
@butchs38 
This is caused by the X-Y hole compensation setting. In the bottom part, it is considered a hole, so it shrinks by 0.2 mm. On the rest of the part, it is not considered a hole; hence, it does not shrink. Because of this change, that layer is indeed an overhang. For this kind of model, X-Y compensation is not recommended.
[image]

butchs38, 2024-06-06:
Thank you that fixed it. One more question on support why is one side different than the other almost hollow. The hollow support always fails. I have been using rec grid which works. Just wondering why it does this.
[image]

closed 2024-09-12 by github-actions[bot]