OrcaSlicer/OrcaSlicer #5566: Object is pasted on top of copied one instead next to it.

santiagomoneta, 2024-06-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 beta

### Operating System (OS)

macOS

### OS Version

14.5 (23F79)

### Printer

Ender 3 v2

### How to reproduce

1. Select an object
2. Keyboard shortcut for COPY 
3. Keyboard shortcut for PASTE

### Actual results

visually there is no action, like nothing happened and no object is pasted next to the copied one, unless you move it to select to auto arrange the plate

### Expected results

as soon I PASTE, the new object should be placed next to the one I originally copied from 

### Project file & Debug log uploads

[Hook_3mm.3mf.zip](https://github.com/user-attachments/files/15533488/Hook_3mm.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

no

Nukle0n, 2024-06-05:
I disagree, if you want a lot of lot the same object you can use the clone feature by right clicking and specifying how many clones you want.  For my workflow it fits much better that a copy appears exactly where the original did so when editing an object in the slicer to say, remove a hole by copying other parts of the object to cover it, that they're easier to align.

TheLostPrints, 2024-06-11:
The clone feature has the same result. Cloned objects are superimposed over the original object. Visually, it appears as if nothing has happen other that the pop-up warning that the cloned object(s) are too close the the original object.

Nukle0n, 2024-06-12:
> The clone feature has the same result. Cloned objects are superimposed over the original object. Visually, it appears as if nothing has happen other that the pop-up warning that the cloned object(s) are too close the the original object.

I tested it again and it would seem there is indeed an error with clones right now. 1st clone appears next to the original object, then the remainder spawn as overlapping entities.

Afuhualuanshi, 2024-06-29:
Whether pasting or cloning, the newly added model will be stacked in its original position. When can this issue be fixed? The latest version of Bambu Studio does not have this issue. I hope the author can provide an option to choose whether to enable this feature or not

closed 2024-10-05 by github-actions[bot]