OrcaSlicer/OrcaSlicer #5567: Print fails with error "Move out of range: 29.916 1.367 331.092 [-2.000]"

bbinnard, 2024-06-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-Beta

### Operating System (OS)

Windows

### OS Version

Windows11/64 Ver 23H2

### Additional system information

No performance problem - everything runs OK with PrusaSlicer. My printer is running the current version of Klipper.

### Printer

FLSun SuperRacer

### How to reproduce

1.Start print

### Actual results

Print fails with error "Move out of range: 29.916 1.367 331.092 [-2.000]" immediately

### Expected results

Print completes

### Project file & Debug log uploads

[disk.zip](https://github.com/user-attachments/files/15534814/disk.zip)
[klippy.log](https://github.com/user-attachments/files/15534815/klippy.log)
[debug_Mon_Jun_03_06_27_58_6056.log.zip](https://github.com/user-attachments/files/15535266/debug_Mon_Jun_03_06_27_58_6056.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

This is the 3rd time I'm reporting this problem. I received no useful result on prior 2 reports. Everything works OK when I slice with PrusaSlicer. For this trial I used the Orca definition for the new SuperRacer printer type and tried to print with Printable Area set to 300.0001 (which I think came from the Orca definition for my FLSun SuperRacer printer) and also with Printable Area set to 265 which was suggested by a Facebook user. 

My guess is there is some interaction between Klipper and Orca, but I've got no idea what that might be.

SoftFever, 2024-06-03:
Can you share the .3mf project file as well as the .gcode file?

bbinnard, 2024-06-03:
The print file is an STL file, not 3MF.
[disk.zip](https://github.com/user-attachments/files/15536033/disk.zip)
I can recreate it as a 3MF if you need that.

bbinnard, 2024-06-11:
This problem still exists with Ver. 2.1.0-beta. The error message is Move out of range: 29.916 1.367 331.092 [-2.000]. No changes were made to printer since I opened this issue.

SoftFever, 2024-06-11:
Please share the project file(3mf file with settings).

bbinnard, 2024-06-11:
[disk.zip](https://github.com/user-attachments/files/15792752/disk.zip)

I don't know what settings you refer to - I don't think Orca has a way to export it's internal settings.

SoftFever, 2024-06-12:
You can export to a 3mf project in OrcaSlicer file menu

bbinnard, 2024-06-12:
1. The disk.zip file I uploaded above does contain the Orca 3MF file with the test part I tried to print.

bbinnard, 2024-07-08:
Oh well, I guess I'll stick with FLSunSlicer.  Or maybe go back to PrusaSlicer.

closed 2024-10-15 by github-actions[bot]