OrcaSlicer/OrcaSlicer #5573: Minimum layer time: decrease acceleration before slowing down print speed

dewi-ny-je, 2024-06-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Acceleration impacts print quality, especially in modern printers which push input shaping too much and get quite softer corners.

When layers are too fast, some choice has to be made about slowing them down.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would offer the user the option to reduce acceleration first, down to a minimum value, and then reduce print speed. 

This way, at least print quality and details improve if a layer has to be slowed down.

### Describe alternatives you've considered

Potential issue is only that print quality varies along the height.

Ataraxia-Mechanica, 2024-06-04:
Check this out:
https://github.com/SoftFever/OrcaSlicer/pull/5148

dewi-ny-je, 2024-06-04:
> Check this out: #5148

Another option to meet the layer time.

igiannakas, 2024-06-04:
I'm not a fan of decreasing acceleration to solve for this. I think it will cause more artefacts than it's trying to solve. Specifically:
1. The right PA value changes with different accelerations (eg: https://www.reddit.com/r/klippers/comments/tqgv9n/big_difference_in_pressure_advance_value_by/) so varying acceleration especially for walls and surfaces will cause variances in extrusion consistency especially around corners. 
2. The rate of speed change being different (due to different acceleration) will cause uniformity issues in external surface "shine" - some areas will reach higher speeds quicker than others so they will end up more shiny

In summary my recommendation is to tune for consistency - the below need to be as equal as possible
1. Wall speed - affects extrusion consistency
2. Acceleration - affects PA consistency
3. Fan speed - affects shrinkage - higher fan speed makes the area underneath the nozzle shrink at different rates
4. Layer time - affects shrinkage as the toolhead has more or less time to dwell over a specific area

For 1,2 & 3 it is easy to solve -> set your external wall speed to a low value and it should be fairly consistent in appearance. Also my PR above aims to keep the external wall speed static irrespective of layer time (but right now it barely works)

For 4 it's a difficult problem to solve. Ideally we would want some layer time smoothing across layers - run the current algorithm twice - first to calculate the layer time across each one individually, then secondly, where layer time changes happen beyond a certain deviation, reduce the speed of the faster layers to match the speed of the slower layers within a deviation margin. This will result in smoothing out the layer time changes so they are not as abrupt between two adjacent layers.

Number 4 could be a game changer but it will be very tricky to implement I think in the current slicer code structure. This has been discussed before in Prusa's GitHub repo here:
https://github.com/prusa3d/PrusaSlicer/issues/2785
https://github.com/prusa3d/PrusaSlicer/issues/8629
https://github.com/prusa3d/PrusaSlicer/issues/9009

@SoftFever it would be awesome to have a chat around point (4) above as this can be a good way to solve for some tricky print artefacts caused by radical speed deviations between layers. Would love your thoughts on how something like this may be able to be implemented.

igiannakas, 2024-06-05:
@SoftFever So I've done some tests on this:

**Left equal layer time - Right equal speeds**
[image]

**"Equal" layer time:**
Layer time:
[image]

Speed used:
[image]

The layer time for each layer was between 43.1 sec and 44.1 sec in the blue areas after the screw stopped printing so a deviation of approx 2%. Where there is a solid layer (green area) the layer time was 47.2 sec, so a deviation of 9%.

**Equal speeds:**
Layer time:
[image]
Speed:
[image]

**Honestly I am puzzled.**
1. The equal layer time cube even though it was printed much slower overall (speeds of around 18mm/sec) turned out looking far worse than the equal speeds cube (that was printed at 140 & 250 mm/sec external and internal perimeters). That I did not expect as I thought that printing slower and with less variation in layer time and speed it would turn out better. But nope...
2. Where there is a sharp increase in layer time (the orange band in the middle of the print) in the equal speeds cube I see no or very minuscule artefacts.
3. It probably looks like equal speeds between layers is the bigger lever to reduce slight cross layer variation.
 
More experimentation is needed. I'm printing another cube with a slower perimeter (50mm/sec) and equal speeds to compare to the faster equal speeds cube. As I need to rule out that for some reason the Voron likes to print faster :P 

If it turns out that equal speeds matter the most then the original PR I've got open is probably the way to go - try to limit speed changes as much as possible caused by variations in slowdown due to layer time not being met.

dewi-ny-je, 2024-06-05:
I forgot to mention that my idea of reducing acceleration would make sense not with technical parts but with organic models which are more acceleration limited, and where layer lines tend to disappear more.

I'm not sure it changes anything, maybe other solutions are better, I mentioned it for completeness.

igiannakas, 2024-06-05:
Hm... Now I am puzzled.

**Left to right:**
Equal layer time (and slow speeds overall) - Middle (equal speeds with slower external walls @50mm/sec) - Right (equal speeds with faster walls @ 140mm/sec)
[image]

Looks like the faster print has less artefacts :S And that I cannot explain...

igiannakas, 2024-06-05:
I'm moving the conversation to the PR so it doesnt pollute this issue - https://github.com/SoftFever/OrcaSlicer/pull/5148

closed 2024-06-16 by SoftFever