OrcaSlicer/OrcaSlicer #5577: The world coordinates of objects are chaotic.

qyr001, 2024-06-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

e.g. 2.0

### Operating System (OS)

Windows

### OS Version

Windows 7/8/10/11

### Additional system information

cpu: Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz   3.19 GHz
mem: 16.0 GB
NVIDIA GT610

### Printer

Bambu Lab X1 ,    Bambu Lab P1P,   KINGROON KLP1.

### How to reproduce

1, The world coordinates of moving objects are very chaotic, making it difficult to place objects correctly through coordinates.
2, The gcode file cannot display a preview image when uploaded to the printer.(printer:   KINGROON KLP1).
3, When the Z-coordinate of the object is 0, it cannot be placed correctly on the base plate.

### Actual results

[image]

### Expected results

When the Z-coordinate of the object is 0, it cannot be placed correctly on the base plate.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15543150/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2024-06-05:
Pretty duplicate to https://github.com/SoftFever/OrcaSlicer/issues/5315
and related closely to https://github.com/SoftFever/OrcaSlicer/issues/5313

The number of times people open new bugs for the coordinate system should be an indicator of how much people want it fixed.

BlackZero500, 2024-06-05:
Also i might add, if you change those coordinates, lets say in this case X Coordinate from -2.7 to 0, it wouldnt go to zero. I cant really see the logic in it but i get the feeling its like a pendulum. In my experience it goes to 2.6, i input 0 again and it goes to -2.4 and so on. i gotta input 0 about 20 times or so until i get it to zero.

tlhintoq, 2024-09-05:
> I cant really see the logic in it but i get the feeling its like a pendulum. In my experience it goes to 2.6, i input 0 again and it goes to -2.4 and so on. i gotta input 0 about 20 times or so until i get it to zero.

I'm pretty sure its measuring from the center of the object in all 3 axis.
[image]

tlhintoq, 2024-12-08:
Not stale
We are all despirate for some kind of coordinate system overhaul that makes sense.
Nothing crazy.  Just nice if an object placed at 0 was on the bed.
And if object 1 is 50mm tall, and object 2 is placed at 50.2mm when you have a .2 layer height then object 2 should be sitting on top of object 1.

This business with the location being the center in all axis is a nightmare

jryan15, 2024-12-24:
Please, please, please consider to make z=0 put the bottom of the part on the bed.  I just finished a large project that required stacking many objects and using color blending on the top of each surface, and it was a nightmare trying to get parts to land exactly on top of other parts.  Even knowing that z was at the middle of the layer was incredibly difficult.

Also when you start moving objects from within assemblies, it is even harder to track.

Please try to develop an option to put z=0 as bottom of part on the bed.  I would happily contribute for this improvement :-)!

tlhintoq, 2025-03-25:
Still an issue that NUMEROUS people would like to see resolved and is the subject of multiple tickets with various wordings but all boiling down to the same thing.

tlhintoq, 2025-03-25:
Haha... Here's one that was the very next email update.
https://github.com/SoftFever/OrcaSlicer/issues/5315

tlhintoq, 2025-06-24:
Still needs attention.  Simply trying to left justify 5 objects is painful. Because you can't put them all at X=50 for example, if they are different sizes.

Plus... What's the point or use of `Object Coordinates`?  It doesn't even seem selectable.  Or maybe only under certain conditions but no idea what they are.  Just seems broken.

https://github.com/user-attachments/assets/de331be1-389d-4989-b242-ebcd7332e9bb

PhilBaz, 2025-08-22:
Just wanted to link these issues as conceptually they suffer from the same problem...nonsensical word coordinate issues. This area of the software need a rework to bring consistency and usability.

https://github.com/SoftFever/OrcaSlicer/issues/10484

*** From the perspective of the user, having the center of the object (some arbitrary z value above ground) Serve as the Z coordinate center makes almost no sense for actions like scaling, and moving relative to the build plate and other objects. The Z-coordinate of the object should be at the plate Z=0.