OrcaSlicer/OrcaSlicer #5578: fatal error: Access violation / object has zero size - if part is mirrored

sledwina, 2024-06-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro

### Additional system information

Prozessor	AMD Ryzen 9 5950X 16-Core Processor               3.40 GHz
RAM	64,0 GB
AMD Radeon VII
Nvidia RTX 3070 Ti

### Printer

Voron 2.4

### How to reproduce

Import DDBDOuterNeckRight.stl - Clone the part - Mirror along x or y axis - hit "arrange" button - 
gives error "object xxx.stl has zero size and can't be arranged"

[image]

Or:
Import DDBDOuterNeckRight.stl - Clone the part - Mirror around x or y axis - rotate part to fit on the build plate - hit "slice plate" - gives the error "fatal error: Access violation"

[image]

### Actual results

See above...

### Expected results

Mirror a part and slice without errors.

### Project file & Debug log uploads

[DDBDOuterNeckRight.zip](https://github.com/user-attachments/files/15550375/DDBDOuterNeckRight.zip)
[debug_Tue_Jun_04_12_55_12_27044.log.zip](https://github.com/user-attachments/files/15550400/debug_Tue_Jun_04_12_55_12_27044.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

GoldenGorilla123, 2024-06-07:
I've got a similar issue, but I can clone, mirror(by x axis) and arrange, but when I go to slice it I get 'access violation' 
[image]

Mirroring by y-axis seems to work around the issue for me temporarily

2.1.0-beta
Windows 11

Intel i7-1185g7
16gb RAM
Integrated Graphics

Voron 2.4 300mm

[Project file.zip](https://github.com/user-attachments/files/15722376/Project.file.zip)
[debug_Fri_Jun_07_09_03_38_7692.log.zip](https://github.com/user-attachments/files/15722379/debug_Fri_Jun_07_09_03_38_7692.log.zip)

daijoubu, 2024-06-10:
Same issue here.  Selecting "Fix Model" before arranging seems to solve the issue with arranging, but it still raises an exception when slicing.

2.1.0-beta
Windows 11
AMD Ryzen 7 5800X3D
32GB Ram
Nvidia RTX 4070

Creality K1C rooted.

[HornetMountMirrored.zip](https://github.com/user-attachments/files/15756078/HornetMountMirrored.zip)
[debug_Sat_Jun_08_14_28_37_6216.log.zip](https://github.com/user-attachments/files/15756111/debug_Sat_Jun_08_14_28_37_6216.log.zip)

mattddonnelly, 2024-06-16:
I'm also having this issue, with the attached file.

I can slice just fine with one object, or with two objects, but if I mirror either one or both objects, I get an access violation error and am unable to slice.

2.1.0-beta
Windows 11
AMD Ryzen 9 5900X 12-Core Processor               4.20 GHz
Installed RAM 64.0 GB
Nvidia RTX 3080

[clamp_handle.zip](https://github.com/user-attachments/files/15856028/clamp_handle.zip)

sledwina, 2024-06-18:
For me the issue is fixed with the 2.1.0 RC.

PR #5827 (2024-06-26): Fix few mirror related issues
Fixes bugs like #5578, #4719 by not using `get_scaling_factor`
Fix  #5578, Fix #4719

Issue can be easily reproduced by:
[image]
closed 2024-06-18 by sledwina