OrcaSlicer/OrcaSlicer #5581: The filament weight counter

Maelow, 2024-06-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The filament weight counter is based on slicer calculations

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The counter of the used filament and the possibility to reset it. A counter that summed up how much filament was used when printing models. Take the slicer calculation as the basis of the calculation and sum it up after sending the job to the printer and subtract it in case of cancellation

This would make it possible to predict the remaining thread on the spool

Maelow, 2024-06-06:
> What you're looking for is spool man. Take a look here: https://github.com/Donkie/Spoolman

It requires a server. There is no integration with OrcaSlicer i with how much filament is sent for printing

igiannakas, 2024-06-07:
It’s integrated through klipper I believe? At least the ERCF I’m using now offers that possibility.

Maelow, 2024-06-08:
I use bambulab, there is no clipper here

closed 2024-09-14 by github-actions[bot]