OrcaSlicer/OrcaSlicer #5582: Time cost not included in the costing in slicing process

zero-one-soft, 2024-06-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

windows 11 22631.3672

### Additional system information

Processor	12th Gen Intel(R) Core(TM) i5-12400   2.50 GHz
Installed RAM	32,0 GB (31,8 GB usable)
System type	64-bit operating system, x64-based processor

### Printer

Creality K1 

### How to reproduce

Enter a value in the time cost field in the printer settings and the slice a model . Change the value to something else and then slice again and the costing will stay the same. This issue was not present in the 2.0.0 stable version 

### Actual results

The value of the costing in the slicing estimate section stays the same when the time cost value was changed

### Expected results

Fix the time cost calculation so that it is included into the estimation once the model was sliced

### Project file & Debug log uploads

[debug_Tue_Jun_04_11_47_53_10860.log.zip](https://github.com/user-attachments/files/15552427/debug_Tue_Jun_04_11_47_53_10860.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

smwoodward, 2024-06-05:
It's the same on Mac OS

Sethi85, 2024-06-05:
same here. it doesn't add on total cost.
i'm on windows 11.

SergeyKolomoitsev, 2024-06-11:
same here. Win 11, Doesn't add the cost of time working. Only adds the filament costs.

Exerqtor, 2024-06-11:
Win10, same issue..

PR #5698 (2024-06-15): Fix time cost not being considered in total cost
Fixes #5582
Fixes #5643
closed 2024-06-15 by SoftFever