OrcaSlicer/OrcaSlicer #5584: Support drawing with straight lines

sparky-summer, 2024-06-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Drawing straight, even lines to block support on edges is hard and kinda slow process.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be useful, if you can paint supports with straight lines, by clicking on a start point, then shift-click where the end point should be, just like in Photoshop. Maybe this is not too hard to add.

tlhintoq, 2024-06-05:
Another common UIX method for this is holding the `Shift` key while dragging to limit movement to straight up/down, left right.

A stop-gap way to help this would be to provide the `[x] Vertical` option from Seam painting to the other tools like support painting.  There's a level of UIX consistency with this approach.
[image]

The same grid overlay idea to help visual straight horizontal and vertical movements would help here as much as for seams.
https://github.com/SoftFever/OrcaSlicer/issues/5464

Vantage-DS, 2024-06-07:
This can be done already, but it's not a "feature".
1) Press P to open your seam painter, click where you want the seam to be, but don't release your mouse button.
2) Press P again to close the seam painter
3) Release the mouse button.
4) Press P again to open the seam painter - Click where you want the end to be..
5) A line is magically drawn between the 2 locations.

Takes 2 seconds!

The only problem is that, if you are zoomed in and the model takes the full screen, you must have the start and end point in view, otherwise where ever you click on the model after re-opening the seam painter, it'll draw that line.... i.e you can't click and drag to move the camera.

tlhintoq, 2024-06-08:
> This can be done already, but it's not a "feature".
>
> 1. Press P to open your seam painter, click where you want the seam to be, but don't release your mouse button.
> 2. Press P again to close the [...]

Holy cr•p... Really?  Wow.  Is this documented someplace/anyplace?  Or is it some weird side-effect that nobody has caught and it just so happens, that it happens?  Some kind of happy-accident caused by the code routine for painting still running even when the paint dialog gets closed.

Vantage-DS, 2024-06-08:
Definitely a happy accident 😄, there's no documentation that I'm aware of. I found the info in a Reddit thread quite a while ago for Prusa Slicer, and it still works in Bambu and Orca Slicer!

tlhintoq, 2024-06-08:
Having written code for drawing and mouse tracking in the past (20+ years in software)...
Once I started doing it by your instructions I could practically see how this is working as far as code for monitoring `mouse-down, location x,y` then `mouse up location x,y` - This is a wonderful happy accident and saved me so much time while getting me a really nice seam over some changing topography.
[image]

closed 2024-09-15 by github-actions[bot]