OrcaSlicer/OrcaSlicer #5587: Options for tree supports disable for any style except 'organic'

tlhintoq, 2024-06-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-dev nightly 05jun2024

### Operating System (OS)

Windows

### OS Version

Win10x65

### Additional system information

11th gen i9... 64gig ram, Nvidia

### Printer

all

### How to reproduce

Select support type Tree.  Style "organic" 
You can change the values of `First Layer Density` and `First Layer Expansion`.
Change to another style of tree such as "heavy".
These options become disabled.  Why?

### Actual results

[image]

### Expected results

These options should be availabe regardless of tree support style.  Organic, light or heavy doesn't change your need to have a bigger brim around the base of the tree support in certain cases.

### Project file & Debug log uploads

Unrelated to any specific project or model.

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2024-06-05:
Note: kinda related to https://github.com/SoftFever/OrcaSlicer/issues/4476 
So maybe someone looking at one might see related to the other bug and can tidy them both up at the same time.

sneakyjoeru, 2024-07-12:
This needs to be fixed

tlhintoq, 2024-10-11:
Don't close it - still not fixed

tlhintoq, 2024-12-29:
While someone is looking at organic supports  maybe they can see why they break up so badly.
As soon as you try to use them on anything of larger size they get these massive rips in them.
[image]

tlhintoq, 2025-03-30:
Please fix support calculation.  There's at least 10 different tickets for similar/related issue of the supports having these gapping holes that are almost as if the support itself layer-shifted in the code.

[image]

sneakyjoeru, 2025-03-30:
It will literally try to print in the air now

[image]

[image]

[image]

[image]

[image]

[image]

[image]

tlhintoq, 2025-03-30:
@sneakyjoeru 
> It will literally try to print in the air now

That's not just "now".  Its been doing mid-air, unsupport supports for a long time.  I've reported them for a long time and I have to guess they were happening before I was reporting them.

I saw a comment on one of the multiple tickets for support problems about this being an inherited issue from Prusa slicer.  So if the Orca people can't attend to this maybe we need to find or create this same problem in Prusa and bang on them.

sneakyjoeru, 2025-06-29:
Let's keep it open

tlhintoq, 2025-07-11:
nudge

tlhintoq, 2026-10-30:
{bot nudge}