OrcaSlicer/OrcaSlicer #5591: Link to printer setting  in error message doesn't work

genuineted, 2024-06-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-beta

### Operating System (OS)

macOS

### OS Version

macOS v14.5

### Additional system information

- CPU: 2.3 GHz 8-Core Intel Core i9
- Graphics: AMD Radeon Pro 5500M 8 GB, Intel UHD Graphics 630 1536 MB
- Memory: 32 GB 2667 MHz DDR4

### Printer

AnyCubic Kobra Plus

### How to reproduce

1. Opened a model.
2. Error message opens: 
> Error:
> Relative extruder addressing requires resetting the extruder position at each layer to prevent loss of floating point accuracy. Add "G92 E 0" to layer _gcode.
> [jump to (before_layer_change_gcode)]
3. click the link provided in error message.

### Actual results

Nothing happens at step 3.

### Expected results

Clicking link in error message should open Printer Settings > Machine G-code (Advanced) and scroll to "Before layer change G-code" section.

### Project file & Debug log uploads

[Baseball Stand.3mf.zip](https://github.com/user-attachments/files/15573718/Baseball.Stand.3mf.zip)
[debug_Tue_Jun_04_22_00_39_75233.log.0.zip](https://github.com/user-attachments/files/15573728/debug_Tue_Jun_04_22_00_39_75233.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

When I did figure out how to open the printer's Machine G-code, I saw this in the "Before layer change G-code" section:
`;BEFORE_LAYER_CHANGE
;G92 E0.0
;[layer_z]
`
The recommendations in the error message were already inserted into the field. One difference, though: The field shows 1E0.0` and the error message recommends `E0`

closed 2024-09-12 by github-actions[bot]

dlannan, 2025-03-16:
Im actually getting this error on every single stl I load in the latest V2.3.0-rc. 
On windows 11. 
Ive tried previous version as well (V2.2.0) and its the same problem. If I use Cura its fine.
The fact all my stl's were all working before is bizarre. I dont understand why the slicing process would change.

WeirdConstructor, 2025-06-04:
Same issue here, I just downloaded the newest version 2.3.0 ( OrcaSlicer_Linux_AppImage_V2.3.0.AppImage ), setup my AnyCubic Vyper from the startup wizard and loaded an STL file. And I get the message "Relative extruder addressing requires resetting the extruder position at each layer to prevent loss of floating point accuracy. Add "G92 E0" to layer_gcode.

The Link in the error message does not work. Please reopen the issue or link to the solved issue, if it is a duplicate.

My questions as user:
- Why do I have to do that?
- Why do I have to do that manually? (Add button to the message, for adding it).
- Why is that not default?
- Is there an issue with the AnyCubic Vyper profile?