OrcaSlicer/OrcaSlicer #5597: 3DConnexion dialog blocked by MainFrame eating "Cmd-Shift-M" improperly on Mac

SuperTango, 2024-06-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 (and all previous versions)

### Operating System (OS)

macOS

### OS Version

macOS Ventura 13.3.1

### Additional system information

2019 MacBookPro 64GB RAM, 4TB SSD (2.4 GHz 8-Core Intel Core i9, AMD Radeon Pro 5500M 8 GB, Intel UHD Graphics 630 1536 MB)

3dconnexion Spacemouse Wireless, 3DxMacCore version 1.3.3.446 (446), 3DxWare version 10.8.3 (3682)

### Printer

Bambulab P1S (but bug is printer independent)

### How to reproduce

* Open OrcaSlicer and go to the "Prepare" or "Preview" tab
* hit "Cmd-Shift-M"

### Actual results

The Orca Slicer screen minimizes into the MacOS Dock

### Expected results

The OrcaSlicer "3DConnexion " dialog should appear.

This happens for all projects (or even when no project is loaded).

### Project file & Debug log uploads

No project file is needed

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

The problem is that hitting "Cmd-Shift-M" on mac always minimizes the app, even on the "Prepare" and "Preview" screen (where Cmd-Shift-M should bring up the 3Dconnexion preferences dialog.)

I tried all of the mac system based ways to get OrcaSlicer to ignore "Cmd-Shift-M" like the suggestions here: https://apple.stackexchange.com/questions/115562/how-do-i-disable-the-minimize-command-m-shortcut, but none of them worked because OrcaSlicer manages all of the keyboard shortcuts on its own so the OS is not involved with handling the "minimize" keyboard shortcut (as far as i can tell).

I'm providing a pull request that fixes the problem:

The code that minimizes (using the WXWidgets "Iconize()" call) happens in MainFrame.cpp keyboard event loop. The code that's checking, looks for "Cmd-M" but does not check for any other keyboard modifiers (such as the Shift key), so I added a check to ignore the event if Shift is pressed along with "Cmd-M".

There's a secondary issue that isn't really relevant to this bug in that the app will still minimize when pressing "Cmd-Shift-M", but ONLY on the "Home" sub-screen. (all other sub-screens work as they should).

I'm not sure why, but when the "Home" sub-screen is selected, the keyboard event loop (MainFrame.cpp, line 546), is called TWICE when "Cmd-Shift-" is pressed:

* Once where the event's wxKeyModifier (retrieved via evt.GetModifiers() is set to wxMOD_CONTROL AND wxMOD_SHIFT. (this is correct)
* Once where the event's wxKeyModifier is ONLY set to wxMOD_CONTROL (this is wrong).

Again, this double-event (with the wrong modifiers) only happens when the user is on the "Home" sub-screen. For the context of this bug the 3DConnexion preferences dialog isn't needed on the "Home" sub-screen so this secondary bug doesn't matter. But it does make the UX odd where Cmd-Shift-M will minimize the app when the user is viewing the "Home" sub-screen, but not minimize the app when the user is viewing any other sub-screen.

SuperTango, 2024-06-05:
Created Pull Request #5598 to address this issue

discip, 2024-06-10:
@SuperTango
Since your PR has been merged, I suppose this can be closed, right? 😊

SuperTango, 2024-06-10:
Yup.  Thanks.  I've closed it, but I'm not sure if that's the correct procedure.  reopen it or whatever if i wasn't supposed to close it myself.

closed 2024-06-10 by SuperTango