OrcaSlicer/OrcaSlicer #5604: 2.1.0 Beta Crashes on Startup MacOS (M1)

tripp396, 2024-06-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 beta

### Operating System (OS)

macOS

### OS Version

Sonoma 14.5

### Additional system information

Chip: Apple M1 Pro
Memory: 32 GB

### Printer

N/A

### How to reproduce

1. Open OrcaSlicer
2. Crash

### Actual results

App does not open, get dialog box that says OrcaSlicer quit unexpectedly. Reopen, report, ignore

### Expected results

App opens as normal

### Project file & Debug log uploads

I've pasted what I assume are the relevant lines of the logs. I had to pull out some lines as there is identifying information in the file paths. The other lines complained of issues finding the parent for certain profiles and unable to load certain 3mf files from various locations.

[warning]	2024-06-05 20:22:06.694911[Thread 0x0000000206670c00]:get_network_function, can not find function bambu_network_del_subscribe
[warning]	2024-06-05 20:22:06.695123[Thread 0x0000000206670c00]:get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]	2024-06-05 20:22:06.695178[Thread 0x0000000206670c00]:get_network_function, can not find function bambu_network_get_profile_3mf
[error]	2024-06-05 20:22:07.242412[Thread 0x0000000206670c00]:3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 0x0005): tried: '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file, not in dyld cache)
[warning]	2024-06-05 20:22:08.094805[Thread 0x0000000206670c00]:20:22:08: Navigation request to 'file:///Applications/OrcaSlicer.app/Contents/Resources/web/homepage/index.html?lang=en_US' (target='')
[warning]	2024-06-05 20:22:08.110277[Thread 0x0000000206670c00]:20:22:08: Navigation complete; url='file:///Applications/OrcaSlicer.app/Contents/Resources/web/homepage/index.html?lang=en_US'
[warning]	2024-06-05 20:22:08.147132[Thread 0x0000000206670c00]:20:22:08: Script message received; value = {"sequence_id":1717636928,"command":"get_login_info"}, handler = wx
[warning]	2024-06-05 20:22:08.147300[Thread 0x0000000206670c00]:20:22:08: Script message received; value = {"sequence_id":1717636928,"command":"get_recent_projects"}, handler = wx
[warning]	2024-06-05 20:22:08.182012[Thread 0x0000000206670c00]:20:22:08: Script message received; value = {"sequence_id":1717636928,"command":"modelmall_model_advise_get"}, handler = wx
[warning]	2024-06-05 20:22:08.185283[Thread 0x0000000206670c00]:20:22:08: Document loaded; url='file:///Applications/OrcaSlicer.app/Contents/Resources/web/homepage/index.html?lang=en_US'

### Checklist of files to include

- [X] Log file
- [ ] Project file

tripp396, 2024-06-06:
I've also attempted the newest nightly and the same issue persists

root670, 2024-06-07:
I have this issue as well

tripp396, 2024-06-07:
Unsure if it helps but it also crashes on my windows machine:

[warning]	2024-06-07 07:59:31.538447[Thread 0x00000704]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]	2024-06-07 07:59:31.538447[Thread 0x00000704]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_profile_3mf

those are two of the lines, the other lines are errors that it can't find certain models (I assume to show on the first screen so you can select a recent project and go)

tripp396, 2024-06-07:
I was able to fix this by downloading the 2.0 release, opening OrcaSlicer, letting it install the network plugin, close out of orcaslicer completely and install the 2.1.0 beta. Then everything worked!

Will close this out but wanted the resolution noted somewhere.

closed 2024-06-07 by tripp396