OrcaSlicer/OrcaSlicer #5605: in device tab can't download files from klipper

BigC4, 2024-06-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

i am unable to download files such as log_files or time_lapses from the device_tab, having to open a web_browser to do so witch makes it pointless to use.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Allow downloading files from device_tab in klipper to pc like any normal browser

### Describe alternatives you've considered

use a Normal web browser to download files

### Additional context

it's a waste of resources on the pc and 3D_printer so why bother using the device_tab if you can't do such a basic task.

dankdeals, 2024-08-02:
having the same issue across all machines. please fix and or allow using default system browser other than edge

GrzesMis, 2024-08-18:
How to fix it?

closed 2024-11-24 by github-actions[bot]