OrcaSlicer/OrcaSlicer #5606: Max Fan setting.

Cblaze401, 2024-06-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 beta

### Operating System (OS)

Windows

### OS Version

Win 10

### Printer

Ender 3 v3 KE

### How to reproduce

Set max fan speed to 40%
Fan runs at 73% during print. 
[image]
[image]

### Actual results

Fan runs a 73%

### Expected results

Fan should run at 40% other than overhangs and bridges 

### Project file & Debug log uploads

[pool_tool_handle_1.zip](https://github.com/user-attachments/files/15600982/pool_tool_handle_1.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

buzzhuzz, 2024-06-06:
Attaching faulty gcode would help understanding the problem.

Cblaze401, 2024-06-08:
> Attaching faulty gcode would help understanding the problem.

![image](https://github.co
[Pool-Tool_Handle_PETG_1h55m.gcode-202457-235553.zip](https://github.com/user-attachments/files/15746374/Pool-Tool_Handle_PETG_1h55m.gcode-202457-235553.zip)
m/SoftFever/OrcaSlicer/assets/154857527/67868a19-5dcf-4557-8e9c-f7fbb6f06352)

buzzhuzz, 2024-06-08:
This gcode looks good: M106 is present whree it should be. Most of the part is prented with "M106 S102" which is matches 40% speed.

The problem may be related to the printer UI. Could you try to close and re-open printer's page several time along the print?

kylek29, 2024-06-17:
Just to note, I came across a related (or same?) bug. The UI elements maybe represented in %, but the Slicer doesn't convert them to 0-255 format **IF** they're emitted via placeholders in one of the various G-Code boxes (I tested with Machine G-Code). 

Example:

- Aux Fan is set to 85% in the UI
- Expected: GCode generated should output: `M106 P2 S216;`
- Actual --> On Machine G-Code Box `
	M106 P2 S{additional_cooling_fan_speed[initial_no_support_extruder]} ; ` --> slicer generates --> `M106 P2 S85;` 

This seems to be because the Slicer is outputting it in the GCode header (variable placeholder storage) as:

`; additional_cooling_fan_speed = 85`

This is on the 2.0.0 (official release).

closed 2024-09-24 by github-actions[bot]