OrcaSlicer/OrcaSlicer #5610: 2.1.0 Beta linux appimage cannot SEND job to printers

emelbardis, 2024-06-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Linux

### OS Version

Fedora 40

### Printer

Bambu Lab X1C

### How to reproduce

Open from appimage
slice plate or plates
click SEND and select printer
errors on 'check lan'

### Actual results

Job doesn't send and errors on check lan

### Expected results

job is sent to printers sd card for printing later 

### Project file & Debug log uploads

[Mini+Fruit+Crates.zip](https://github.com/user-attachments/files/15627908/Mini%2BFruit%2BCrates.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

This has been present for a month or so on nightly builds and I had hoped the 2.1.0 beta would fix. Am currently stuck on 2.0 official

mrehkopf, 2024-06-08:
I have a similar issue, but in my case the "Send to printer" dialog simply "almost freezes" (becomes very unresponsive for 1-2 minutes, never shows a printer) before I can close it again. Printer (Bambu X1 Carbon) is reachable and controllable fine from the "Device" tab.

mrehkopf, 2024-06-08:
In my case, logging out of my Bambu account and re-logging in just did the trick, so it was probably unrelated.

closed 2024-10-28 by github-actions[bot]