OrcaSlicer/OrcaSlicer #5612: Camera Feed not loading - Bambu Lab P1P, Beta 2.1

awelch702, 2024-06-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

i9-13900k
DDR 5 64 Gigs
4070 TI Graphics

### Printer

Bambu Lab P1P

### How to reproduce

Click Device Tab
Hit Play on the Camera
Error: "Please check the network and try again. You can restart or update the printer if the issue persists. [2:-10]
[image]

### Actual results

[image]
Error - The Printer is online. When I revert to stable 2.0 there is no issue.

### Expected results

Camera to display

### Project file & Debug log uploads

[debug_Thu_Jun_06_12_31_36_346932.log.zip](https://github.com/user-attachments/files/15685603/debug_Thu_Jun_06_12_31_36_346932.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

N/A

2lm, 2024-06-06:
I have exactly the same problem with Beta 2.1 and P1P/P1S in Windows 11.

cochcoder, 2024-06-07:
Try the latest nightly version, it should be fixed now

eschreiner, 2024-06-07:
> I got a pop up asking if i wanted to leverage the bambusource.dll but it doesn't appear to have fixed it. Tried to uninstall and reinstall to get a screenshot of it but it does not seem to come back. I noticed the nightly was posted 10 hours ago, so I will try with tomorrows nightly release

Same for me. Reverting back to 2.0.0 fixed the video stream. I'm using a P1S with latest firmware.

awelch702, 2024-06-07:
The nightly build released this morning resolved it

Blawlz, 2024-06-08:
Just posting to say the nightly build from July 8 release fixed this for my P1S as well. 
Took two installation attempts of the build though. The first installation gave me a similar pop up about the bambusource.dll and when I clicked "yes" it was still broken. Installed the same exe a second time, and the issue is resolved now.

marcin-l, 2024-06-10:
Had the same situation, had to install nightly twice to fix the issue.

Thanks @Blawlz for the tip.

mydevpeeps, 2024-07-04:
still issue, workaround: https://github.com/SoftFever/OrcaSlicer/issues/5754#issuecomment-2209553656

closed 2024-10-11 by github-actions[bot]