OrcaSlicer/OrcaSlicer #5613: Expose "Model Name" to be used for Filename Format variable

RoryCodes, 2024-06-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

no, it would just help me (and possibly others) who would like to use that piece of information when naming their exported Gcode files.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to be able to use the "{model_name}" variable in the Filename Format section to export my gcode accordingly.

example: I open a 3MF file in Orca Slicer called _"super cool widget.3mf"_. The name _"super cool widget"_ populates the "Model Name" field in the Project Tab. Then just before slicing, I populate the Filename Format field with: _"{model_name} - V1"_. I slice the file and export to gcode. the gcode file should be called _"super cool widget - V1"_.

CiclopeEstrabico, 2024-06-13:
I would like to see this happen!

jbjhjm, 2024-06-21:
yes please. This was the default behavior for prusa/super slicer. 
All of my 3mf/gcode files use identical naming for easy identification.
Hard to believe the most basic export naming option is now impossible. :/

jbjhjm, 2024-09-20:
shut up bot! Valid enhancement request!

closed 2024-12-28 by github-actions[bot]

jbjhjm, 2025-01-06:
really? Secretly staling and closing this over christmas? That's f***ed up shit 😡

lordmorgul, 2025-02-08:
I want this too!