OrcaSlicer/OrcaSlicer #5619: Partially covered settings tabs on high resolution screens

gaaat98, 2024-06-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Dell XPS 15 9530, 3456x2160p at 250% scaling (default by windows)

### Printer

n/a

### How to reproduce

Just launch Orca at the specified resolution and scaling

### Actual results

[image]

Clearly a minor problem, yet the settings sections are covered by the surrounding panes, they are still clickable but it requires more precision

### Expected results

The highlighted part should be clearly spaced, visible and easily clickable as they were in version 2.0.0

### Project file & Debug log uploads

[debug_Fri_Jun_07_12_12_45_10488.log](https://github.com/user-attachments/files/15737796/debug_Fri_Jun_07_12_12_45_10488.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

HaernCalistinson, 2024-06-08:
I'm seeing the same thing.

[image]

gaaat98, 2024-06-25:
Fixed in 2.1.0 release

closed 2024-06-25 by gaaat98