OrcaSlicer/OrcaSlicer #5621: Seams are inconsistent when painted, do not aligned properly, and change when multiple (cloned) parts are added to bed.

Vantage-DS, 2024-06-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 (all versions)

### Operating System (OS)

macOS, Windows

### OS Version

N/A

### Additional system information

N/A

### Printer

P1S - Though, irrelevant

### How to reproduce

1) Slice model.
2) Seams are not adequately placed for cosmetic look of final print.
3) Paint seams manually, as nearest/aligned or other options do not give good enough results.

### Actual results

Seams are notoriously difficult to place manually on objects that do not have perfectly straight edges.
Results of seams, even after painting, do not line up or properly cling to sharp edges when using "nearest", nor do they align neatly when using "aligned".

Seam painting takes a very long time on some of my models. One example where i blocked seams everywhere on the model to try get neat results:
[image]

The result is still random:
[image]

Blocking and repainting the tiniest parts of a seam line are the only way to fix this:
Seam close up-
[image]

Fully zoomed in, the painted view of ^that seam painting looks like-
[image]

The minute changes i have to make to make the seam align properly-
[image]

The new result of ^that minute change-
[image]

The problem is further exaggerated when cloning a part.

The original seam painting (same screenshot from earlier):

[image]

The actual seam when 2 objects are now placed on the bed:

Not only does the seam layout from the 'single' model change it's location, which means when i have 2 clones on a plate, the seam needs refining yet again, the seams are also being put where they are supposed to be blocked-
[image]

### Expected results

An aligned/nearest seam should cling to an edge in a straight and neat line like this:
[image]

The seam should stay in that location regardless of how many objects are on the build plate.

The computation of nearest or aligned needs be refined to actually align.

OrcaSlicer is brilliant, and this has been a problem way before OrcaSlicer existed, but i think it's time seams were fixed before i lose my sanity 🙃

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Vantage-DS, 2024-06-10:
More problems....

Switching from my P1S to a Prusa MK4, using near identical slicing settings (other than speed).

This tiny blue dot is where it want's to put the seam (left side of picture)-
[image]

This is the painting. There is no reason for it-
[image]

I try go overboard, and give the slicer more than enough room to add a seam to the correct area, just for testing-
[image]

This is the result. The blue dot/seam still wants to be on the left-
[image]

I've got 3 sizes of this object, and each time i need to slice a variation (1 on the bed, 2 on the bed, 4 on the bed, or even now, a different printer) the seams need refining. It's taking so much time and i have no idea why it's being like this.

NVNDO, 2024-08-13:
Seams in OS and BS are horrible. I don't understand why its so hard to implement this properly like in Cura. 

90% of time I have to draw the seam myself and not even then, like mentioned here it as it should, as the software randomly bends or cut the seam for no reason.

Vantage-DS, 2024-08-13:
Yeah, Cura has great seam placement. I've never painted seams on Cura, because I never needed to 😄

adripo, 2025-02-02:
Hi everyone, I'm still running into the seam alignment issues mentioned here. I’d be happy to help. whether that's testing patches or diving into the code. Could someone point out which part of the logic might need extra attention? I'd appreciate any guidance so we can get this resolved together.

NVNDO, 2025-02-02:
Well.. that I call showing initiative @adripo - I hope when can provide the necessary information and you can sort this out. It would be a game-changer! 🙏🏼

kbroasterc, 2025-02-06:
This and curved seams when set to back has me pulling my hair out, I love to have a fix for this

silvernull, 2025-02-11:
This issue constantly plagues me also.

[image]

Vantage-DS, 2025-04-25:
Simple straight line - Both Aligned and Near:

[image]
[image]
[image]

There must be some code from Prusa Slicer that can be pulled.

I drew some really lazy seams in Prusa (about 10 seconds of work) and it lined up perfectly..... The alternative in Orca Slicer took me about 20 minutes to fine tune, and even then it wasn't guaranteed.

johyphenel, 2025-06-12:
+1, trying to paint a seam to be aligned and in a corner feels impossible with Orca.

M-Smeets, 2026-10-31:
2.5.0 Still valid issue.

NVNDO, 2026-10-31:
It has been a while since i last used Orca, fixing this might make me reconsider it as an option again. (Among other things) 

I sincerely don't understand why it's so difficult to make a logic that joins seam-points instead of placing them sort of random.