OrcaSlicer/OrcaSlicer #5626: View option without perspective

ft276, 2024-06-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently Orcaslicer integrates a form of perspective when viewing the model in Prepare or Preview tabs. I don't understand the reason for this - I never find it helpful, just something to tolerate, and there are times where it is distinctly unhelpful.
For example, I have an object that is narrower at the base than the top, and I would like to make sure I position it correctly on the  build plate. The obvious way to check this is to go to the top-down view. But this centres the 'camera' instead of providing a true top-down view, so the apparent model position changes depending on where the build plate is positioned on the screen.

It's incredibly frustrating, as I am left guessing where the camera might actually consider the centre of the image and the only part with a true top-down view.

Can this be fixed, or an option given to view the objects without perspective added?

Thanks

[image]
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

For the view in the Prepare and Preview panels to simply provide a view in the orientation requested, instead of modelled around a 'central camera' hovering above the object that sees things from its own perspective.

### Describe alternatives you've considered

1. Moving the build plate around until I estimate I have positioned it directly below the camera, for the specific part of the model I am looking at, at this particular moment.
- highly unsatisfactory

jhlchu, 2024-06-08:
Switch view type to `Orthogonal View`
[image]

ft276, 2024-06-08:
Thank you so much! 
Looked all through the settings and didn't think to check that

discip, 2024-06-10:
@ft276
As this issue seems to be solved, please close this. 😊

closed 2024-06-10 by ft276