OrcaSlicer/OrcaSlicer #5635: error while loading shared libraries: libwebkit2gtk-4.0.so.37

ffries, 2024-06-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Debian testing

### Printer

Ender3 S1

### How to reproduce

./OrcaSlicer_Linux_V2.0.0.AppImage 
/tmp/.mount_OrcaSlcXDMEU/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Actual results

AppImage does not start

### Expected results

AppImage should lanch

### Project file & Debug log uploads

N.S.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

libwebkit2gtk-4.0.so.37 cannot install and was removed from Debian.

Unsatisfied dependencies:
 libavif15 : Dépend: libsvtav1enc1 (>= 1.2.1+dfsg) mais il n'est pas installable
Error: Impossible de corriger les problèmes, des paquets défectueux sont en mode « garder en l'état ».

dpoulson, 2024-06-09:
I'm getting the same on a newly upgraded ubuntu 24.04. 

Same error on 2.0.0 and 2.1.0 beta. 

`daz@dogbert:~/Programs$ ./OrcaSlicer_Linux_V2.1.0-beta.AppImage 
/tmp/.mount_OrcaSlgxgF9x/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
daz@dogbert:~/Programs$ ./OrcaSlicer_Linux_V2.0.0.AppImage 
/tmp/.mount_OrcaSlCXOxWd/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
`

dpoulson, 2024-06-09:
Latest Bamboo Studio also has the same issue

`daz@dogbert:~/Programs$ ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage 
/tmp/.mount_Bambu_2CoX8Q/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
`

dpoulson, 2024-06-09:
Bambustudio ticket: 

[https://github.com/bambulab/BambuStudio/issues/3973](https://github.com/bambulab/BambuStudio/issues/3973)

doom-fr, 2024-06-17:
This works for me and is the easier to handle : 
https://github.com/bambulab/BambuStudio/issues/3973#issuecomment-2097651206

ffries, 2024-06-20:
All these issue are related :
https://github.com/SoftFever/OrcaSlicer/issues?q=libwebkit2gtk-4.0.so.37

ccoupe, 2024-08-26:
This happens on a new Mint 22 install and download of OrcaSlicer 2.1.1.  In looking at the other issues this was reported I did'nt see any workarounds - libwebgtk-4.0 is not found be apt.

eingemaischt, 2024-10-12:
> Latest Bamboo Studio also has the same issue
> 
> `daz@dogbert:~/Programs$ ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage /tmp/.mount_Bambu_2CoX8Q/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory `

use flatpak for bambu.

I can't really believe that I can't download a working OrcaSlicer version vor over 7 months, even though someone even created a flatpak version but nobody seems to officially asks for integration to flatpak hub...

closed 2025-01-19 by github-actions[bot]