OrcaSlicer/OrcaSlicer #5641: Holes in certain thin walls

qwirty123, 2024-06-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Window 10

### Printer

Bambu Lab X1 Carbon

### How to reproduce

I stumbled across this in this model of a spiral, the there are holes in the thin walls. This happens whether with step files or stls. Interestingly, it only happens with the model on plate 1 and not on a nearly identical model on plate 2. I used default presets. See the attached file.

### Actual results

There is an oddly shaped hole in one of the thin walls.
[image]

### Expected results

There should be no hole.

### Project file & Debug log uploads

[bug_report.zip](https://github.com/user-attachments/files/15749004/bug_report.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

discip, 2024-06-10:
@qwirty123
This is not caused by the slicer as your stp/stl is corrupted:

[image]

Please close this issue since this is not related to OrcaSlicer. 😊

qwirty123, 2024-06-10:
@discip 
Interesting... Could you see the attached step file and see if it is corrupted? My concern is that there is an issue with the conversion from step to mesh. It is straight from fusion360. Yet when I export as mesh from f360, it is perfectly fine.
[d2_11.zip](https://github.com/user-attachments/files/15774648/d2_11.zip)

qwirty123, 2024-06-11:
There seems to be a bug in the conversion from step to mesh. When you use another tool to convert from step to stl, there is no corrupted mesh.

discip, 2024-06-11:
You don't need to convert to stl.
Orca can use step / stp files as well.

qwirty123, 2024-06-11:
That's what I'm saying. If you use the step file, there are holes; but if you convert the step file to stl first and use that, there are no holes. Thus, there must be something wrong with the way orca handles step files that causes holes.

foreachthing, 2024-06-11:
I've seen "corrupt" STEP files from Fusion360 here as well. When importing the STEP file into SolidWorks and export them again, everything was fine. So, yeah, Orcas STEP import has some room for improvement (specifically for STEP files from Fusion360).

discip, 2024-06-11:
> That's what I'm saying.

Ah, I see!
Sorry, I thought you meant you used another method to convert to stl. 🤦‍♂️

discip, 2024-06-11:
> Please close this issue since this is not related to OrcaSlicer. 😊

My previous comment was made prematurely, as far as I understand after further investigation, and thus is simply wrong.

I imported the original step file (https://github.com/SoftFever/OrcaSlicer/issues/5641#issuecomment-2158500819) into **`F r e e C A D`**, **`c a t i a`** and **`o n s h a p e`**.
The result in the later two was flawless, whereas the first one showed the same artifacts as OrcaSlicer does.

discip, 2024-06-11:
> Please close this issue since this is not related to OrcaSlicer. 😊

My previous comment was made prematurely, as far as I understand after further investigation, and thus is simply wrong. 😓

I imported the original step file (https://github.com/SoftFever/OrcaSlicer/issues/5641#issuecomment-2158500819) into **`F r e e C A D`**, **`c a t i a`** and **`o n s h a p e`**.
The result in the later two was flawless, whereas the first one showed the same artifacts as OrcaSlicer does.

So this indeed seems to be an issue with OrcaSlicer! 😯

qwirty123, 2024-06-11:
It appears that this goes deeper than Orcaslicer as the same artifacts appear in Prusaslicer and Freecad. It's possible that it's a bug with the CAD kernel itself. On the [Prusaslicer release notes for STEP support](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-beta1), it says

> We use [Open CASCADE Technology (OCCT)](https://github.com/Open-Cascade-SAS/OCCT) development platform to read the STEP files. It is a CAD kernel also used e.g. by [FreeCAD](https://www.freecadweb.org/) or [KiCad](https://www.kicad.org/). Thanks https://github.com/Open-Cascade-SAS for keeping it open-source. The import implementation itself was ported from [BambuStudio](https://github.com/bambulab/BambuStudio), thanks https://github.com/bambulab.

I don't have kicad installed, but I suspect it would display the same issues.

discip, 2024-06-12:
@SoftFever
@Noisyfox
@igiannakas
Have you experienced similar behavior?
Is the observation that the kernel is not performing as expected correct?

discip, 2024-07-24:
@SoftFever
@Noisyfox
Anyone? 😊

closed 2024-10-30 by github-actions[bot]