OrcaSlicer/OrcaSlicer #5645: Fan speed not changing/wrong [SPEED] variable for Klipper in gcode?

Ergonomicmike, 2024-06-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-beta portable nightly build

### Operating System (OS)

Windows

### OS Version

Win10

### Printer

Klipperized Ender 3 v2

### How to reproduce

Import my project file.
Slice.
View Fan Speed in Color Scheme and note that fan speed is supposed to change during the print.
View the gcode and search for M106.
[image]

### Actual results

Fan speed stays at 50% for the actual print.

### Expected results

Fan speed should change, as high as 70%.

### Project file & Debug log uploads

[TickRemover_M._Fan Speed Issue Orca.zip](https://github.com/user-attachments/files/15750546/TickRemover_M._Fan.Speed.Issue.Orca.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

So, I might a total idiot. If so, please close this issue.
(But  https://github.com/SoftFever/OrcaSlicer/issues/5606 might be related?)
In the gcode, I see mostly M106 S127. 
Although this is a valid number for Marlin, it doesn't change in my gcode with the height of the print as it should per the Fan Speed Preview of my print. So that's my first issue.
But over at [Kllipper documentation](https://github.com/SoftFever/OrcaSlicer/issues/5606), they say that the Fan Speed value is supposed to be between 0.0 and 1.0. If corect, then a second issue.

discip, 2024-06-10:
No issue here:

[image]

Ergonomicmike, 2024-06-10:
@discip Thanks for checking. I reloaded the project and now my M106's are changing.

Arrgh. I don't know why it's working now. 

But closing this issue.

closed 2024-06-10 by Ergonomicmike