OrcaSlicer/OrcaSlicer #5646: Gaps in perimeters.

user32rtu, 2024-06-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

v2.1.0-beta, v2.1.0-dev, v2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro 22H2

### Additional system information

AMD FX(tm)-8150 Eight-Core Processor  3.60 GHz
16,0 ГБ

### Printer

Sapphire Pro

### How to reproduce

1. Open a project from an attachment.

### Actual results

Gaps appear in some external and internal perimeters.
In places where there are breaks in the G code, the filament is retracted.

### Expected results

Perimeters should not have unpredictable discontinuities.

### Project file & Debug log uploads

[bolt.zip](https://github.com/user-attachments/files/15752470/bolt.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]
[image]
[image]
[image]

thhdragon, 2024-06-10:
the issue disappears if you uncheck retraction and detraction speed in the filament profile overrides. doesnt seem to like having a really low speed like 4.5mm/s

user32rtu, 2024-06-10:
> the issue disappears if you uncheck retraction and detraction speed in the filament profile overrides. doesnt seem to like having a really low speed like 4.5mm/s

Yes, adjusting the retraction speed has an effect on this problem. Some breaks disappear, others don't.

user32rtu, 2024-06-27:
When you set the "Use relative E distances" parameter in the printer profile, the gaps disappear.

closed 2024-10-03 by github-actions[bot]