OrcaSlicer/OrcaSlicer #5650: Painting doesn't flow inward, it just paints top layers

pennystockss, 2024-06-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 Beta

### Operating System (OS)

macOS

### OS Version

14.4.1

### Additional system information

2021 Macbook M1 Pro 14"

### Printer

Any

### How to reproduce

Paint an object

### Actual results

Paints only top surfaces

### Expected results

Paint to slice like previous version

### Project file & Debug log uploads

[Text.3mf.zip](https://github.com/user-attachments/files/15753534/Text.3mf.zip)
[debug_Sun_Jun_09_12_45_48_60886.log.0.zip](https://github.com/user-attachments/files/15753541/debug_Sun_Jun_09_12_45_48_60886.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]
[image]
[image]

igiannakas, 2024-06-09:
Do you have shrink compensation in your filament settings enabled?

pennystockss, 2024-06-09:
Negative. Used default profile and default Bambu PLA filament settings.

tlhintoq, 2024-06-11:
How did you paint?  What painting tool?  If you paint with the `Height Range` tool it paints full layer for me.
[image]

If you paint with the circle tool, then you're only painting the exterior surface.

pennystockss, 2024-06-13:
Yes I did the height range in that example, I assumed it paints like regular just in flat lines across. Here's an astronaut example. Slices fine in 2.0. Circle I assume is the same as 2.0, paint in a flat circle with whatever pixels are on your screen not on the model like the sphere but it should still fill to the inside. Mine's only slicing the top most layer, on all existing projects and if there isn't any top layer, it skips painting all together.

[image]
[image]

pennystockss, 2024-06-13:
Here's in 2.0. Exact same project. Are you on Mac? I wonder if it's a Mac thing, surely if it were everyone it would get a lot more attention.
[image]

pennystockss, 2024-06-22:
Bump, 2.1.0 official still does it.
[image]
[image]

igiannakas, 2024-06-23:
Looks like we may have found the issue - if you build the binaries locally the issue doesn’t occur. So it’s somewhere in the GitHub build chain that produces the release binary.

fix hopefully coming soon as we kind of know where the issue now comes from.

azocarf, 2024-06-24:
same problem  en orca for Mac

igiannakas, 2024-06-25:
Latest Mac build should solve the non shrinkage related MMU issue. Download from the nightly and use this command to "sign" it so it opens.

`xattr -dr com.apple.quarantine /Applications/OrcaSlicer.app`

pennystockss, 2024-06-26:
I grabbed the latest one, didn't need command this time (tried before on another build that did). And it works!

jthomeo, 2024-08-29:
Hi guys. Has anyone managed to solve the problem? I'm using Orca Slicer 2.1.1 and I'm facing the same issue. Is there a work around?

[image]
[image]

igiannakas, 2024-08-29:
Yes - 2.2 nightly solved this I believe. This is the PR that was merged yesterday.

closed 2024-12-05 by github-actions[bot]