OrcaSlicer/OrcaSlicer #5653: Slicing behavior

SolberLight, 2024-06-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Elegoo Neptune 4 Plus

### How to reproduce

Using the given file slicer with normal 1.6 layer height

### Actual results

Walls aren't perfectly straight it's adding geometry from nowhere

### Expected results

Straight walls

### Project file & Debug log uploads

[A4p.zip](https://github.com/user-attachments/files/15763817/A4p.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

foreachthing, 2024-06-10:
Disable Arc-fitting will help. I don't you need that anyway. If you do need it, do it in post with ArcWelder.

discip, 2024-06-10:
@SolberLight

Try this:
[A4p.3mf.txt](https://github.com/user-attachments/files/15777645/A4p.3mf.txt)

If you print it laying flat on the bed, the thin fins will definitely break off.
Hence the angled position, just in case you wonder.

SolberLight, 2024-06-10:
Thanks for your suggestion, actually i've already printed many of these and they infact aren't straight due to the slicing.
thins are plenty strong so not the issue there.

The issue I'm seeing is that Orca is adding material where it shouldn't

[image]

SolberLight, 2024-06-10:
Can be seen here :
[image]

SolberLight, 2024-06-10:
I've tried your settings. It's clearly better, but not perfect, what should I changed next ?
[image]

discip, 2024-06-10:
What is the diameter of you nozzle 0.4 mm or 0.6 mm ... ?

discip, 2024-06-10:
What is the diameter of your nozzle 0.4 mm or 0.6 mm ...?

discip, 2024-06-10:
What is the diameter of your nozzle? 0.4 mm or 0.6 mm ...?

SolberLight, 2024-06-10:
.4 yes

discip, 2024-06-10:
Is it possible to attach the whole file?

discip, 2024-06-10:
Is it possible to attach the whole project file?

discip, 2024-06-10:
Try reducing layer height to at least 1.3 mm.

discip, 2024-06-10:
Try reducing layer height to at least 0.13 mm.

discip, 2024-06-10:
Try reducing layer height to at least 0.13 mm.
Preferably 0.1 mm.

SolberLight, 2024-06-10:
That's a good point that should help. Still there should be a warning or something in orca in this case. print shouldn't start without advice

discip, 2024-09-09:
@SolberLight
Have you tried the latest commit?
Maybe your issue has already been solved . 😊

closed 2024-09-09 by SolberLight