OrcaSlicer/OrcaSlicer #5661: source-aligned spiral z-hop for print quality

czlr, 2024-06-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently, **spiral** z-hops are only done via G3 (counterclockwise arc move) and aligned based on the movement **destination**

For example, referring to the image below:
- (1) -- Initial wall printing and wipe. The z-hop starts from point A
- (2) -- z-hop spiral, ending at the same x,y but at a higher z
- (3) -- straight line motion to the destination point B
  - Currently, the spiral path (2) is generated such that line (3) is tangent to its endpoint. Basically, (2) is aligned to smoothly transition into (3).

**Problem:** I believe, based on theory and testing, that the transition between lines (1) and (2) -- i.e., how the nozzle **begins** the z-hop as it leaves the printing plane -- has a much bigger impact on print quality than the transition between lines (2) and (3). Since by the time the latter happens, the nozzle is already several mm above the print surface.

[image]

The screenshot above is a relatively non-problematic example. But I've had situations where the initial G3 motion juts straight out perpendicular to an outer wall. Which can leave pretty nasty scarring for min-max'ed edge cases where you don't have room to dial other settings like cooling and retraction (fully-dense glass-like PETG, flexible/tough filaments, etc).

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

My theory/proposal is that it would be better for spiral z-hops to be aligned WRT its starting context. I.e., the lines it was just printing. Such that:

- In general, z-hops are intentionally aligned with the direction of the preceding motion
  - In the example picture, this is **tangent** and **moving forward**
- z-hops can vary between G2 or G3 as needed to always curve inwards **towards the filled body**. This is mostly relevant for outer walls.
  - Note: so the perspective gets reversed for inner hole walls, doubly reversed for inner bodies, etc.

[image]

### Describe alternatives you've considered

Re: the specific alignment strategy, I think there are some potential options, with different potential theories for why they could be ideal.
- **A:** curve forward, tangent to initial motion
  - I have confirmed (via custom gcode postprocessor script) that this gives more consistent prints for me. Currently, this is part of my default workflow for prints. I haven't had time to compare this to the other options below though.
- **B:** Slightly angled inwards to reduce wall artifacts further. Similar idea to existing settings: [Wipe on loops] / [Wipe before external loop]
- **C:** Directly inwards 90 degrees. Should in theory minimize external stringing, but might cause artifacts to poke out on the other side for super thin walls/flexures that are just 2 outer walls side-by-side.
- **D:** Wipe backwards. Seems counterintuitive at first glance, but might be beneficial since the most recently printed area is also the softest/hottest. So it might help in absorbing any stringing and reducing the lateral forces/resistance imparted onto the model during z-hop (maintain bed adhesion / less risk of knocking the model during z-hop).

[image]

### Additional context

**Environment:** OrcaSlicer v2.0.0 + Bambu P1S

Nathan22211, 2024-06-23:
I feel like we need some more testing than just one of your printers. You said you had a postscript. Maybe if you provided that we could confirm this to be a common thing with spiral Z hop.

czlr, 2024-06-29:
Apologies for the delay. Since posting, I might've found a better approach -- rather than aligning with the current motion, just do a static align vs the current layer's walls.

E.g.,
1. suppose purple is our initial printing path, and we want to zhop at the blue dot
2. look for the closest wall, and align the center of the zhop spiral (I and J params) so that it wipes away from the wall
3. choose either G2 or G3 for the spiraling to better match the current path motion. looking inwards from the wall, if the path is more "left to right", use G3 (case 1 in pic). otherwise use G2 (case 2)

[image]

Here's a postprocessing script I threw together to test
- Currently using Python 3.12.3 on windows, 3rd party modules are listed separately: (frozendict, aenum, numpy, shapely)
- On Orca, just put the script directly into the postprocessor without any args. You can also run the script directly on a single gcode file, in which case it tries to be smart and doesn't overwrite the original.
- On line 34 you can try experimenting with different relative angles. I was also experimenting a bit with the effects of adding a slight pause, but don't have any conclusive results on that yet.
- To get the gcode changes showing up on Orca preview and then passed to the printer, I also have to run the steps mentioned here (https://forum.bambulab.com/t/post-processing-scripts/25517/12). Basically, export both 3mf and processed gcode after slicing, unzip the 3mf, replace the gcode + md5 checksum file, repack, then open in orca. (Would appreciate if anyone happens to know a better way for this)

https://gist.github.com/czlr/a7859730299e73ac01c4d85a8a82827d

[image]

closed 2024-10-05 by github-actions[bot]

CodeMasterCody3D, 2024-10-08:
Send it to me I'll see if I can incorporate it to a build. Here is my alternative email. codydixonipad@gmail.com