OrcaSlicer/OrcaSlicer #5662: Dovetail Width Too Narrow

JoshuaLyle, 2024-06-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

12.6.7

### Printer

FLSun V400

### How to reproduce

Cut a dovetail in a very tall object nearly the full width of the object
[image]

Attempt to cut an equal length dovetail on a different part of the still-too-long part of the object but the width is unable to match the original width.
[image]

### Actual results

The width of a second dovetail cut is unable to match the width of the first dovetail cut when the object was complete

### Expected results

The width of the dovetail should be able to match or exceed the full width of the object. It seems it is current restricted by some ratio of the maximum Z height.

A really odd workaround is to scale the object put to 200% on the z axis, then I'm able to make the cut with the expected width, then scale the objects back down o 50%.

### Project file & Debug log uploads

[dovetail problem example.3mf.zip](https://github.com/user-attachments/files/15780530/dovetail.problem.example.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I need to make a very wide dovetail to split rectangle in my project to cut around the middle of an oval but only the first oval can be split. The second oval's dovetail cut is unable to go wide enough and it cannot be entered manually.

discip, 2024-06-11:
You are right, for some reason the max cut width seem to decrease with every other cut.

Maybe the height is taken into the calculation as well? 🤷‍♂️

JoshuaLyle, 2024-06-11:
interestingly, it seems to just be related to the largest dimension because even if you rotate it 90 degrees so that the super-long axis is sitting flat, the max width does not change

JoshuaLyle, 2024-06-15:
Well for anyone who is experiencing this issue and needs an immediate fix, I found a better way to get around it for now.

Make a compound object that is disconnected from your main object along the cut axis (or maybe whatever is the longest dimension)?) and put it very far away. Then make a compound object in whatever your CAD software is so it'll be imported into Orca as a single object. This is how to do it in FreeCAD
[image]

Then make the first cut like normal and you should have access to plenty of width to be able to do this
[image]

Then the added height from the compound object tricks orca into giving you more width to cut with so you can make a second equal-width cut
[image]

Then just make a planar slice of the floating extension object in the compound and it can be deleted so that you can proceed as normal
[image]

Not the most elegant thing but it works for now

closed 2024-09-21 by github-actions[bot]