OrcaSlicer/OrcaSlicer #5664: Slicing issue colour loss

eriko411, 2024-06-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1beta

### Operating System (OS)

macOS

### OS Version

Monterey 12.6.8

### Printer

Bambu a1

### How to reproduce

[image]
Skull has black eyes for multi colour but this is removed when slicing 

### Actual results

[image]

### Expected results

Eyes should stay black

### Project file & Debug log uploads

[orca skull colour issue.3mf.zip](https://github.com/user-attachments/files/15784979/orca.skull.colour.issue.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Vantage-DS, 2024-06-11:
Just noticed this on a project too.
Set base colour to gold in slot 4.
Painted 2/3 upper model blue in slot 1, using height range.

Sliced model only shows absolute top layer as blue.

eriko411, 2024-06-11:
This worked last month only change is no fourth colour on the AMS.

iv another issue with making overhangs printable now messes up the models base it mixes colours into the overhang 🙈

tlhintoq, 2024-06-11:
Maybe related... maybe not...
Bring in a model that's all painted for one printer like Bambu X1C.
Decide you want to print it on the Bambu P1S (totally compatible machine), so you change printers and all the painting is lost.
Maybe something happening to re-load the printer at slice time causing the same loss??

rplanier, 2024-06-16:
Same problem here.  Painting a height range a particular color results in only the top layer actually being sliced with that color.  All other layers lower in the painted range are sliced in the original color.

tlhintoq, 2024-06-16:
Last night's nightly in Windows 10: Paint by layer is correctly painting all layers that color.
[image]

Come down partway through the painted black range... Black all the way through
[image]

Come down partway through the painted gold range... Gold all the way through
[image]

igiannakas, 2024-06-20:
Can you please test this fix here? https://github.com/SoftFever/OrcaSlicer/pull/5774

Maximizer18, 2024-06-23:
I tried the build linked, and still having the same issue as everyone else, and have skrinkage set to 100%.
[image]
[image]
[image]

Maximizer18, 2024-06-23:
And a simple cube, clearly is painted with 4 colors, yet only sliced with 2.
[image]
[image]

igiannakas, 2024-06-24:
Separate issue has been identified with the Mac builds in GitHub. If you build it locally it works but the GitHub build doesn’t. Under investigation.

closed 2024-08-28 by SoftFever