OrcaSlicer/OrcaSlicer #5669: Brim type Mouse ear inside Text facing printbed

Gron-HD, 2024-06-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-beta

### Operating System (OS)

Windows

### OS Version

Win 11 Pro 23H2

### Printer

Ender 3 V2

### How to reproduce

Load any model with a 7mm text size, orient the model so the text pointing to the bed. Select Brim type "Mouse ear".

### Actual results

[image]

### Expected results

There should be a brim type "mouse ear", thats not inside text, but everwhere were it is in the screenshot elsewhere.

### Project file & Debug log uploads

[LCD-holder-brim-bug.zip](https://github.com/user-attachments/files/15793715/LCD-holder-brim-bug.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2024-06-12:
I second that.

TrentBartlem, 2024-09-02:
Same. I'm printing thin rectangular shapes where mouse ears makes the most sense, but brim filling negative text makes the text practically pointless.

discip, 2024-12-02:
Unfortunately still an issue with the latest dev build.

discip, 2025-03-10:
With the latest dev build this is till present.

@Noisyfox
Maybe you could have a look. Also this happens with **`Brim type > Painted`**.

discip, 2025-03-10:
With the latest dev build this is till present.

@Noisyfox
Maybe you could have a look.
This also happens with **`Brim type > Painted`**.

Noisyfox, 2025-03-10:
> This also happens with **`Brim type > Painted`**.

How could this happen if you don't put the brim inside text?

discip, 2025-03-10:
Sorry, I didn't make myself clear enough. 😅

This is the case I was reffering to:

[image]

What I meant was:  
Could we please have more control over the position of the brim, specifically regarding its application to the inside or outside?  

Maybe we can achieve this by adjusting the current behavior as follows:  
1. The **`Brim type`** options **`Mouse ear`** and **`Painted`** would be removed.  
2. As soon as **`Brim Ears`** is used, the Brim type would automatically be set to the current **`Painted`** state but would then be influenced by the setting chosen in **`Brim type`**, such as **`Outer brim only`**.

I hope this clarifies things. If not, please let me know and I’ll be happy to explain further.

Thank you very much in advance!

I appreciate your work very much! 😃👍

Noisyfox, 2025-03-10:
See original discussions here regard fine controls of brim ears https://github.com/SoftFever/OrcaSlicer/pull/1779#discussion_r1292668795

Noisyfox, 2025-03-10:
> This is the case I was reffering to:
> 
> [image]

This issue though, is an efficency compromise. Brims at one side of the polygon could grow into the otherside if the wall is too thin. Though consider how little brims usually are, it might be affordable to introduce a finer detection algorithm to avoid such issue.

discip, 2025-03-10:
Thank you.

> See original discussions here regard fine controls of brim ears [#1779 (comment)](https://github.com/SoftFever/OrcaSlicer/pull/1779#discussion_r1292668795)

Ok, I read that.

> ... though I also don't want to introduce too many settings that could confuse users.

That's exactly my point.

As you know currently you need to use **`Brim Ears`** and additionally set **`Brim type`** to **`Painted`**.

What I suggest is:
Change the current procedure, so that it would suffice to only use **`Brim Ears`** which would cause the second step (setting **`Brim type`** to **`Painted`**) to be applied automatically in the background.
That also means that **`Brim type > Mouse Ears`** & **`Brim type > Painted`** need to be removed, as the function of **`Brim type > Mouse Ears`** is redundant in regard to **`Brim Ears`**.

This should make it easier.

I hope you get my point. 😊

Noisyfox, 2025-03-10:
> That also means that **`Brim type > Mouse Ears`** & **`Brim type > Painted`** need to be removed, as the function of **`Brim type > Mouse Ears`** is redundant in regard to **`Brim Ears`**.

This do look more convient, but that would break existing projects, which I also want to avoid.

discip, 2025-03-10:
Obviously good points. 😊
Yeah, no easy task.

Noisyfox, 2025-03-11:
> This is the case I was reffering to:
> 
> [image]

One other problem is, in this case, if you open the brim painting gizmo, the painted gizmo does cover the internal hole, which, for some user, will expect the brim to appear in there. If we apply a filter so it only generate brims outside, could confuse those users.

closed 2025-06-17 by github-actions[bot]