OrcaSlicer/OrcaSlicer #5673: Add Sovol new "High speed" profile for sv06

Matigon16, 2024-06-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Current processes and .json do not include the high speed profile

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

replace resources/profiles/Sovol and resources/profiles/Sovol.json with these files: 
**Sovol json**
[Sovol.json](https://github.com/user-attachments/files/15796041/Sovol.json) 
**Sovol Folder**
I cannot upload a direct folder, found in the google drive in SV06 High Speed\SV06 Marlin High Speed File\3.Orca Slice Windows\Sovol Profiles
 which are directly from the sovol High speed profile guide in google drive: https://drive.google.com/drive/folders/1nOCIkOwC72kUQ6fCGt0dNmMTQbQu7B7k

### Additional context

Files provided come directly from sovol: https://drive.google.com/drive/folders/1nOCIkOwC72kUQ6fCGt0dNmMTQbQu7B7k that url is found in https://www.sovol3d.com/products/sovol-sv06-best-budget-3d-printer-for-beginner?variant=47245979877685

cochcoder, 2024-06-12:
Thanks for finding this!

closed 2024-06-15 by SoftFever

cyberlussi, 2024-11-02:
In the process of upgrading my local settings for SV06 High-Speed, I compared the Sovol GDrive (as of 2024-11-02) and OrcaSlicer 2.2.0 provided files and I stumbled over that some files specifically the `machine/Sovol SV06 0.4 nozzle.json` has not been updated in OrcaSlicer resources with #5675. When slicing with the new High-Speed profile, OrcaSlicer raises a warning that "The acceleration setting exceeds the printer's maximum acceleration".

I have no experience whether the new Sovol provided speed/acceleration values are good values or marketing values, but IMHO the slicer warning can be avoided by updating (at least) machine_max_acceleration_extruding (Sovol=5000). Maybe the other machine values should be updated as well for the sake of consistency with Sovol values.

PR #5675 (2024-06-15): Add Sovol SV06 High Speed process that Sovol provides
Closes #5673