OrcaSlicer/OrcaSlicer #5679: Nightly Build release notes?

Avpman2, 2024-06-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Where are the release notes for the nightly builds? It's very hard to evaluate the builds when there is no indication what to test. Also, it would be VERY helpful if a minor rev number were added to each build. Thanks!

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Revision notes to be posted.

Blawlz, 2024-06-13:
I would also be interested in locating info for nightly releases. The 2.1.0 Beta did not like the Bambu integrated webcam. The nightly build fixed that, but it appears that the K factor / Factor K setting that used to be present in the AMS screen is broken now or has a partially complete feature. I will open an separate issue for that

TimLovesTech, 2024-06-16:
Not as neat as someone typing them all out, but you can look at the commit history for each build to see what was added/removed.

[Nightly Build Commits](https://github.com/SoftFever/OrcaSlicer/compare/nightly-builds...main)

closed 2024-09-23 by github-actions[bot]