OrcaSlicer/OrcaSlicer #5683: Could not Skip Object: Error References Wrong Version of Bambu Studio

CCS86, 2024-06-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 Beta

### Operating System (OS)

Windows

### OS Version

Win 10 x64

### Printer

X1C

### How to reproduce

I was printing a job with 4 objects, and needed to skip one.

### Actual results

When I clicked the skip button, I got this error:

[image]

### Expected results

In the past I have been able to skip objects on prints sliced by OS.

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

yummypho, 2024-10-07:
I can confirm this issue.  

On a slightly related note:  When I try to initiate a new part print, I notice the wrong part being printed.   I've tried 3+ times to confirm not a user error.   

I'm using OS 2.1.1  My Bambu P1S is 1.06.01, which is currently the latest..

drawgas, 2024-12-24:
I have the same issue. OrcaSlicer 2.2.0, P1S with firmware 1.07. Does not allow through Bambu Handy App (says wrong version) and no option to do that in OrcaSlicers device view as well