OrcaSlicer/OrcaSlicer #5692: Nightly Build: PA Profile / AMS "Factor K" selection or input issue

Blawlz, 2024-06-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 Beta - Nightly Build 6/8/24

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambulabs P1S +AMS

### How to reproduce

1. Device Tab
2. AMS area
3. Click "edit" icon on Filament (any slot, 1 through 4)
[image]

### Actual results

"Factor K" cannot be manually entered as prior versions allowed.
If a calibration is performed, the final calibration page allows input of the K factor and asks for the filament name. Once accepted, this appears to create a "PA Profile" and show the K factor in the AMS screen as before.
[image]
[image]
However, the K factor cannot be changed or manually input and only the previously calibrated profile is available for selection.

### Expected results

The Factor K should be editable for manual input from the AMS slot pop-up. Values for existing PA Profiles should be editable to accommodate quick changes.

I have previously recorded the K factor for my filaments on swatches. Since the PA K factor is the only tuning I perform, I do not generally save each filament separately and use the "Generic" profiles most often.

I do love the idea of having "PA profiles," but it would be beneficial to have an editable list at a minimum to avoid having to run a calibration print to re-create the previously calibrated values.

Any advice on other work arounds is appreciated. Thank you to everyone who contributes to the ongoing improvements of this slicer!

### Project file & Debug log uploads

[PA Profile issue.zip](https://github.com/user-attachments/files/15827298/PA.Profile.issue.zip)
[Debug Log.zip](https://github.com/user-attachments/files/15827308/Debug.Log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

P.S. Because I have not seen this PA Profile option previously, this may be a newly added feature and this bug may be better categorized as a feature request of the new feature. I believe the behavior is more of an issue than a lack of a feature, but reassign the label if needed.

Highlander-bdj, 2024-07-26:
It is exhibiting the same behavior on the Mac ARM build in version 2.1.1.

lglasgow-godaddy, 2024-09-14:
Yes this is sorely needed as the P1S and I assume the P1P only can save 16 Factor K results. I have both and X1C and a P1S so I do the calibration on the X1C and bring over those results to the P1S and it would be nice to have that manual input back as I note those results down like the original author of this enhancement does.
Mac ARM 2.1.1

closed 2024-12-22 by github-actions[bot]