OrcaSlicer/OrcaSlicer #5695: Dark Mode. When exporting settings the checkbox is not visible

sergeyfatman, 2024-06-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU Core I7 13-gen, RAM 64Gb

### Printer

Anycubic Mega S

### How to reproduce

1. go to 'File'
2. go to 'Export'
3. choose 'Export Preset Bundle'
4. Check 'Process presets(.zip)

[image]

### Actual results

[image]

### Expected results

when I move the mouse to this area, a checkbox is showing

[image]

### Project file & Debug log uploads

[debug_Fri_Jun_14_01_17_18_8336.log.zip](https://github.com/user-attachments/files/15829527/debug_Fri_Jun_14_01_17_18_8336.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2024-06-14:
@jamincollins
Could you please have a look?
Since you are quite familiar with this kind of issue after all what you have achieved in this regard. 😊

HAndreasH, 2024-06-21:
noticed same issue on latest 2.0.1-rc (21-06-2024)
all checkboxes in the export dialog in dark mode are not visable

closed 2024-09-28 by github-actions[bot]